wasm-micro-runtime/wasm-micro-runtime #10: WASI support?

Widdershin, 2019-05-10:
Hi 👋 

Excellent initiative, great to see more Wasm runtimes from experienced vendors.

I was hoping to find out what your feelings/plans are around supporting WASI, Mozilla’s proposed systems interface for Wasm applications.

Thanks again for this project 🙂

PerArneng, 2019-05-10:
Agree. Its very important that standards are created at this time. If you build standards great things happen. Wasm itself is proof of that. Wasi is a good idea and its important that more organisations get behind it.

xwang98, 2019-05-10:
@Widdershin @PerArneng , 
Thanks for bringing the WASI to us. We are interested in WASI and agree that it is important for enabling cross platforms support for wasm applications.

There are some technical questions though when we look at the current WASI api definition below:
https://github.com/CraneStation/wasmtime/blob/master/docs/WASI-api.md

It looks the definitions are more targeting file IO for the desktop and server platforms, not for embedded systems. For example, in an embedded controller usually there is no file system.  

Maybe we can defined a profile of WASI for embedded environment?

lukewagner, 2019-05-13:
One of the goals of WASi (described [here](https://hacks.mozilla.org/2019/03/standardizing-wasi-a-webassembly-system-interface/)) is to be modular, with APIs grouped into discrete modules so that a host environment can implement only the modules that make sense on that host.  It's still very early in WASI's development, but I think it's likely that the [WASI core](https://hacks.mozilla.org/2019/03/standardizing-wasi-a-webassembly-system-interface/) will be stripped down so file IO will be in a separate module that embedded environments wouldn't need to implement.

beriberikix, 2019-07-16:
There is a reference implementation of [`wasi-libc`](https://github.com/CraneStation/wasi-libc). I believe it could replace the current subset implementation, based on the methods in common. That might be all that's needed to support WASI. @sunfishcode?

sunfishcode, 2019-07-16:
Are you referring to the libc APIs descriped [here](https://github.com/intel/wasm-micro-runtime/blob/master/core/iwasm/lib/app-libs/libc/lib_base.h)? If so, then yes, WASI libc does contain implementations of all of those functions.

Kevin0626, 2019-07-16:
This project is required to work within 300K memory budget, including the memory used for holding runtime binary. So the footprint and memory usage is a high priority design goal. To save the memory for holding those libc functions, we just selected to export the native string functions to WASM application rather than compiling them into the WASM bytecode.

@sunfishcode Hi Dan, we are working together with Josh. Maybe we can schedule a meeting for sharing the projects and discussing how to collaborate on WASI standard.

xwang98, 2019-11-27:
supported on Linux now.

closed 2019-11-27 by xwang98

beriberikix, 2019-11-28:
While the issue was more general about WASI, is there plans for support to other hosts like Zephyr?

xwang98, 2019-11-28:
The problem for WASI support on Zephyr is that embedded OS like Zephyr doesn't have standard file IO implementation. It is usually project based decision on what fs middleware to be used.

xwang98, 2019-11-28:
@yiting16, do you have any plan to implement the WASI for Vxworks?

beriberikix, 2019-11-28:
Agreed, but as I understand it Zephyr includes a [filesystem abstraction](https://docs.zephyrproject.org/latest/reference/file_system/index.html) which could be leveraged!

xwang98, 2019-11-28:
@beriberikix thanks for sending this reference! we didn't notice that!  we will keep watching it.

yiting16, 2019-11-28:
We are investigating WASI, no concrete plan at this point.