wasm-micro-runtime/wasm-micro-runtime #5: Provide an example/tools on how to compile source code without emscripten.

almosnow, 2019-05-10:
emscripten is basically llvm + a runtime lib for browser

if this project already provides the runtime then there is no need for emscripten in the toolchain

furthermore, the latest llvm already provides some improvements in this regard

I was able to do this some time ago (compile a minimal .wasm program with only llvm) but it was a very tricky endeavour so if other developers want to achieve the same thing they would have to walk through stones to accomplish it

so, a working an example and pipeline for compiling without emscripten would be a nice addition to this repository

xwang98, 2019-05-10:
Thanks for the comments, @almosnow 

You should be right that a compiler like clang should be able to do the work. We will do some tests and update the README accordingly.

xwang98, 2019-06-04:
The issue is solved in PR #37

closed 2019-06-04 by xwang98