wasm-micro-runtime/wasm-micro-runtime #16: JIT compilation ?

sletz, 2019-05-11:
I read "WASM interpreter (AOT is planned)": will it be with a JIT? For intel CPU? ARM CPU ? What kind of performances compared to native code can be expected ?

wenyongh, 2019-05-14:
Thanks for the comments, @sletz. We are planning to enable AOT feature with some codegen library, e.g. LLVM or cranelift. And JIT is also an option, but it is not the main choice for embedded system, since it needs to link to the codegen library, which is large for embedded system.
For the CPU, ARM and x86 should be enabled. For the performance, we will try to make it as fast as we can.

beriberikix, 2019-07-16:
There is at least one AoT compiler from Fastly called `lucetc` that may be helpful. More details can be found [here](https://github.com/fastly/lucet/wiki/lucetc).

darkmoon233, 2019-07-23:
Does wamr currently not use jit?

wenyongh, 2020-02-18:
WAMR JIT and AOT had been enabled.

closed 2020-02-18 by wenyongh