wasm-micro-runtime/wasm-micro-runtime #19: Drop instruction not implemented

vshymanskyy, 2019-05-11:


wenyongh, 2019-05-14:
Thanks vshymanskyy. DROP instruction is converted to DROP_32 or DROP_64 when loading wasm binary, according to the data type in stack, so it is not implemented in interpreter.

closed 2019-05-14 by wenyongh

vshymanskyy, 2019-05-19:
@wenyongh thanks, did not notice it at first.
This leads to another question! Is there a way to pre-process the original wasm file beforehand?

wenyongh, 2019-05-20:
@vshymanskyy, yes, we perform the type safe checking when loading wasm binary, based on a type interpreter, which is to execute the bytecode where operands are types instead of real data values. For example, for i32.add opcode, the type interpreter checks and pops two i32 types from stack, and pushes one i32 type back to stack. And when processing DROP instruction, it checks operand type in stack top, if it is I32 or F32, we connvert DROP to DROP_32, and if it is I64 or F64, we convert DROP to DROP_64.