wasm-micro-runtime/wasm-micro-runtime #20: Failed to build for 64bit target

TerryGuo, 2019-05-12:
The 32bit build is successful. However after changing cmake variable ```BUILD_AS_64BIT_SUPPORT``` from "NO" to "YES" to enable 64bit build, the ```make ```  will fail with below message:
```
[ 24%] Building C object CMakeFiles/vmlib.dir/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c.o
/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c:329:23: error: ‘get_va_list’ declared as function returning an array
 static inline va_list get_va_list(uint32 *args)
                       ^~~~~~~~~~~
/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c: In function ‘get_va_list’:
/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c:336:8: warning: returning ‘__va_list_tag *’ from a function with return type ‘int’ makes integer from pointer without a cast [-Wint-conversion]
 return u.v;
        ^
/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c:336:8: warning: function returns address of local variable [-Wreturn-local-addr]
/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c: In function ‘parse_printf_args’:
/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c:356:12: error: assignment to expression with array type
 *p_va_args = u.v;
            ^
make[2]: *** [CMakeFiles/vmlib.dir/build.make:219: CMakeFiles/vmlib.dir/home/terry/wasm/wasm-micro-runtime/core/iwasm/lib/native/libc/libc_wrapper.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/vmlib.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
I am working on Ubuntu with GCC 8.3.

wenyongh, 2019-05-14:
Thanks a lot, TerryGuo! Now the 64bit build error is fixed, could you please try again? Compiling wamr in 64bit is an experiment feature, please help notify us if there are issues found.

TerryGuo, 2019-05-15:
Confimed. I can now build for 64bit target.

closed 2019-05-15 by TerryGuo