wasm-micro-runtime/wasm-micro-runtime #21: Need an example to show how to use WAMR standalone

TerryGuo, 2019-05-12:
Thanks for the excellent work. I am trying to integrate WAMR into my code to read a WASM file and call fib function which isn't calling any system abi functions. However I couldn't make it work even for below simple code:
```
#include "wasm-export.h"

int main ()
{
	wasm_module_t module;
  	wasm_module_inst_t inst;
  	wasm_function_inst_t func;
  	wasm_exec_env_t env;
  	wasm_runtime_init();
	return 0;
}
```
The error messages are as below:
```
erry@tPad:~/wasm/wasm-micro-runtime/core/iwasm/products/linux/build$ gcc -m32 test.c -I /home/terry/wasm/wasm-micro-runtime/core/iwasm/runtime/include/ -lm -lpthread -lvmlib -liwasm -L /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build
/usr/bin/ld: /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build/libvmlib.a(bh_thread.c.o): in function `_vm_thread_sys_init':
bh_thread.c:(.text._vm_thread_sys_init+0x49): undefined reference to `pthread_key_create'
/usr/bin/ld: bh_thread.c:(.text._vm_thread_sys_init+0xa8): undefined reference to `pthread_key_delete'
/usr/bin/ld: /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build/libvmlib.a(bh_thread.c.o): in function `vm_thread_sys_destroy':
bh_thread.c:(.text.vm_thread_sys_destroy+0x35): undefined reference to `pthread_key_delete'
/usr/bin/ld: /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build/libvmlib.a(bh_thread.c.o): in function `vm_thread_wrapper':
bh_thread.c:(.text.vm_thread_wrapper+0x43): undefined reference to `_bh_log'
/usr/bin/ld: /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build/libvmlib.a(bh_thread.c.o): in function `_vm_thread_create_with_prio':
bh_thread.c:(.text._vm_thread_create_with_prio+0xc9): undefined reference to `pthread_attr_setstacksize'
/usr/bin/ld: bh_thread.c:(.text._vm_thread_create_with_prio+0x145): undefined reference to `pthread_create'
/usr/bin/ld: /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build/libvmlib.a(bh_thread.c.o): in function `_vm_tls_get':
bh_thread.c:(.text._vm_tls_get+0x49): undefined reference to `pthread_getspecific'
/usr/bin/ld: /home/terry/wasm/wasm-micro-runtime/core/iwasm/products/linux/build/libvmlib.a(bh_thread.c.o): in function `_vm_tls_put':
bh_thread.c:(.text._vm_tls_put+0x4c): undefined reference to `pthread_setspecific'
..............................
```

Would you please kindly provide some examples to show how to use WAMR standalone? Thanks very much.

xwang98, 2019-05-13:
@TerryGuo please refer to the source files under {repo root}/core/iwasm/products/linux.  You can build a standalone executable and take wasm binary file as input for execution.

TerryGuo, 2019-05-14:
Got your point. Thanks. This resolved my problem.

closed 2019-05-14 by TerryGuo