wasm-micro-runtime/wasm-micro-runtime #29: No Windows support?

DragonOsman, 2019-05-19:
There's only Zephyr and Linux support.  Someone else already asked about macOS, so here's a Windows one.  

So yeah, are there any plans for supporting Windows?  And how much of the C++ standard library, if any at all, does this already support?

WASI is still a work in progress and it doesn't support C++ yet, but hopefully both projects will do so at some point and work together if possible.

xwang98, 2019-05-21:
@DragonOsman there is no plan to add support Windows now, but we appreciate contributions from community to add Windows support. It should not be difficult.

wenyongh, 2020-09-08:
WAMR interpreter for windows 32b-it and windows 64-bit, and AOT for windows 32-bit are supported now. Seems that we can close issue.

closed 2020-09-08 by wenyongh