wasm-micro-runtime/wasm-micro-runtime #84: Null pointer dereference - remove_tree_node (ems_alloc.c:45)

pventuzelo, 2019-07-23:
### Environment

| Questions            | Answers
|----------------------|--------------------
| Related Binary       | ./iwasm (linux build)
| Commit               | commit 9a02c49d7c2bb2012fe8d2bd2b9cb2bdaf88ba79 

### Vulnerability/issue

Null pointer dereference in remove_tree_node (ems_alloc.c:45)

### Steps to reproduce the behavior 
- Download: 
XXX
- Run:
`./iwasm PoC.wasm`

### Additional Informations

Crash
```
[1]    5683 segmentation fault  ./iwasm 
```

GDB
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.

[----------------------------------registers-----------------------------------]
RAX: 0x4040404040404040 ('@@@@@@@@')
RBX: 0x0 
RCX: 0x0 
RDX: 0x55555577f134 --> 0x6d736100a0000078 
RSI: 0x555555793b37 ('@' <repeats 200 times>...)
RDI: 0x555555793b37 ('@' <repeats 200 times>...)
RBP: 0x7fffffffd6d0 --> 0x7fffffffd710 --> 0x7fffffffd760 --> 0x7fffffffd780 --> 0x7fffffffd7a0 --> 0x7fffffffd7c0 (--> ...)
RSP: 0x7fffffffd6b0 --> 0x7fffffffd710 --> 0x7fffffffd760 --> 0x7fffffffd780 --> 0x7fffffffd7a0 --> 0x7fffffffd7c0 (--> ...)
RIP: 0x5555555748c5 (<remove_tree_node+118>:	mov    rax,QWORD PTR [rax+0x10])
R8 : 0x55555577eeb8 --> 0x1 
R9 : 0x7fffffffd468 --> 0x0 
R10: 0x0 
R11: 0x246 
R12: 0x555555557960 (<_start>:	xor    ebp,ebp)
R13: 0x7fffffffda70 --> 0x2 
R14: 0x0 
R15: 0x0
EFLAGS: 0x10202 (carry parity adjust zero sign trap INTERRUPT direction overflow)
[-------------------------------------code-------------------------------------]
   0x5555555748b8 <remove_tree_node+105>:	call   0x555555557740 <__assert_fail@plt>
   0x5555555748bd <remove_tree_node+110>:	mov    rax,QWORD PTR [rbp-0x18]
   0x5555555748c1 <remove_tree_node+114>:	mov    rax,QWORD PTR [rax+0x18]
=> 0x5555555748c5 <remove_tree_node+118>:	mov    rax,QWORD PTR [rax+0x10]
   0x5555555748c9 <remove_tree_node+122>:	cmp    QWORD PTR [rbp-0x18],rax
   0x5555555748cd <remove_tree_node+126>:	jne    0x5555555748e1 <remove_tree_node+146>
   0x5555555748cf <remove_tree_node+128>:	mov    rax,QWORD PTR [rbp-0x18]
   0x5555555748d3 <remove_tree_node+132>:	mov    rax,QWORD PTR [rax+0x18]
[------------------------------------stack-------------------------------------]
0000| 0x7fffffffd6b0 --> 0x7fffffffd710 --> 0x7fffffffd760 --> 0x7fffffffd780 --> 0x7fffffffd7a0 --> 0x7fffffffd7c0 (--> ...)
0008| 0x7fffffffd6b8 --> 0x555555793b37 ('@' <repeats 200 times>...)
0016| 0x7fffffffd6c0 --> 0x0 
0024| 0x7fffffffd6c8 --> 0x0 
0032| 0x7fffffffd6d0 --> 0x7fffffffd710 --> 0x7fffffffd760 --> 0x7fffffffd780 --> 0x7fffffffd7a0 --> 0x7fffffffd7c0 (--> ...)
0040| 0x7fffffffd6d8 --> 0x555555574c46 (<unlink_hmu+383>:	nop)
0048| 0x7fffffffd6e0 --> 0x555555793b37 ('@' <repeats 200 times>...)
0056| 0x7fffffffd6e8 --> 0x55555577eea0 (0x000055555577eea0)
[------------------------------------------------------------------------------]
Legend: code, data, rodata, value
Stopped reason: SIGSEGV
0x00005555555748c5 in remove_tree_node (p=0x555555793b37 <global_heap_buf+85143>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/ems/ems_alloc.c:45
45	    if (p == p->parent->right) {
#0  0x00005555555748c5 in remove_tree_node (p=0x555555793b37 <global_heap_buf+85143>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/ems/ems_alloc.c:45
#1  0x0000555555574c46 in unlink_hmu (heap=0x55555577eea0 <global_heap_buf>, hmu=0x555555793b37 <global_heap_buf+85143>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/ems/ems_alloc.c:124
#2  0x00005555555757af in gc_free_i_heap (vheap=0x55555577eea0 <global_heap_buf>, obj=0x555555799090 <global_heap_buf+106992>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/ems/ems_alloc.c:507
#3  0x000055555555dc4f in mem_allocator_free (allocator=0x55555577eea0 <global_heap_buf>, ptr=0x555555799090 <global_heap_buf+106992>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/mem_alloc.c:43
#4  0x000055555555ddab in bh_free_internal (ptr=0x555555799090 <global_heap_buf+106992>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/bh_memory.c:134
#5  0x000055555555ddef in bh_free (ptr=0x555555799090 <global_heap_buf+106992>) at XYZ/wasm-micro-runtime/core/shared-lib/mem-alloc/bh_memory.c:283
#6  0x0000555555576510 in wasm_hash_map_destroy (map=0x555555780a68 <global_heap_buf+7112>) at XYZ/wasm-micro-runtime/core/iwasm/runtime/utils/wasm_hashmap.c:289
#7  0x000055555556d420 in wasm_loader_unload (module=0x55555577f1b0 <global_heap_buf+784>) at XYZ/wasm-micro-runtime/core/iwasm/runtime/vmcore-wasm/wasm_loader.c:1496
#8  0x000055555556d161 in wasm_loader_load (buf=0x55555577f138 <global_heap_buf+664> "", size=0x74, error_buf=0x7fffffffd900 "h\242\377\367\377\177", error_buf_size=0x80) at XYZ/wasm-micro-runtime/core/iwasm/runtime/vmcore-wasm/wasm_loader.c:1436
#9  0x00005555555594f5 in wasm_runtime_load (buf=0x55555577f138 <global_heap_buf+664> "", size=0x74, error_buf=0x7fffffffd900 "h\242\377\367\377\177", error_buf_size=0x80) at XYZ/wasm-micro-runtime/core/iwasm/runtime/vmcore-wasm/wasm_runtime.c:137
#10 0x000055555555802d in main (argc=0x1, argv=0x7fffffffda80) at XYZ/wasm-micro-runtime/core/iwasm/products/linux/main.c:196
#11 0x00007ffff7448b97 in __libc_start_main (main=0x555555557d8c <main>, argc=0x2, argv=0x7fffffffda78, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffda68) at ../csu/libc-start.c:310
#12 0x000055555555798a in _start ()

```

Valgrind
```
==5681== Memcheck, a memory error detector
==5681== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==5681== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==5681== Command: ./iwasm XYZ/wasm-micro-runtime/core/iwasm/products/linux/debug/triage/0x555555574962/PoC.wasm
==5681== 
==5681== Conditional jump or move depends on uninitialised value(s)
==5681==    at 0x4C32CF9: strlen (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5681==    by 0x54799D1: puts (ioputs.c:35)
==5681==    by 0x10C04C: main (main.c:198)
==5681== 

==5681== 
==5681== HEAP SUMMARY:
==5681==     in use at exit: 0 bytes in 0 blocks
==5681==   total heap usage: 1 allocs, 1 frees, 1,024 bytes allocated
==5681== 
==5681== All heap blocks were freed -- no leaks are possible
==5681== 
==5681== For counts of detected and suppressed errors, rerun with: -v
==5681== Use --track-origins=yes to see where uninitialised values come from
==5681== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

wenyongh, 2019-07-24:
Hi, pventuzelo, thanks a lot for your help to test and report so many security vulnerabilities. We tested some of the uploaded files and found that they are invalid wasm binary files, or the file format of them is invalid, e.g.:
$ wasm-objdump -s PoC.wasm
  error: function body must end with END opcode
  error: expected valid local type
  error: unable to read float: f32.const value
  error: unable to read string: section name

Actually these files should be rejected by wasm loader and should not be executed by wasm intepreter. Currently we have applied many integrity checks of wasm file in wasm loader, we will apply more in the future and make it more robust and secure, and fix the issues you report.
But it is not the goal of the product currently, we hope that the wasm file integrity is guaranteed by user, and the wasm file provided to wamr is with valid format. For example, user can add signature after generating the wasm file, and check the signature to ensure that it isn't modified before sending it to wamr.

pventuzelo, 2019-08-06:
Hi, 

I understand your point of view but, even if some static checks are done by the wasm loader, some vulnerabilities in the interpreter can still happen since there is no complete validation of the processing data at runtime.

Do your "module signature" ensure that the module is valid or it only ensure that no bytecode modification have been done between compilation and execution?

wenyongh, 2019-08-06:
Hi, yes, thanks a lot, some checks can only be done in interpreter, such as memory boundary check for load/store opcode, table elem index out of range for call_indirect, stack overflow check, integer divided by zero check, and so on. We added many checks in interpreter.
We are testing your issues, and basically fixed them, we will submit the code when test is done.
For module signature, you are right, it only ensures no bytecode modification is done after compilation.

pventuzelo, 2019-08-06:
Ok thanks, let me know when it's pushed on master ;)

wenyongh, 2019-08-08:
Hi pventuzelo, many thanks, we have merged the patch, which should have fixed all the issues you found, could you help confirm it? If there are new issues found, please let us know, we are glad to resolve them.

pventuzelo, 2019-08-08:
Ok good, let me check and i will directly close the tickets ;)

pventuzelo, 2019-08-08:
Good for me ;) 
Thanks for your work !

closed 2019-08-08 by pventuzelo