wasm-micro-runtime/wasm-micro-runtime #85: How to call a native webassembly file in c/c++ source code?

darkmoon233, 2019-07-25:
In the JavaScript, you can use the `WebAssembly.instantiate` function to load a WASM file and use the various methods provided by the WASM file. I want to know if there is such a way in C++ that I can use WASM files directly.
Is there some API between WASM and C/C++?

wenyongh, 2019-07-26:
Yes, you can embed iwasm in C++ to call functions provided by WASM file, though the API's provided by iwasm is written in C,  you can also call them in C++. I uploaded an example:
include folder: iwasm's head files
lib folder: iwasm's lib file libiwasm.so, which can be generated when building iwasm in Linux
src folder: an example of C++ source files, which loads wasm file, instantiates the module, lookup main and fib functions, and then calls them
wasm-src: the source code of wasm file

After running ./build.sh, you can run
  export LD_LIBRARY_PATH=$PWD/lib
  ./iwasm test.wasm
to see the output results.

[iwasm_cpp.zip](https://github.com/intel/wasm-micro-runtime/files/3434115/iwasm_cpp.zip)

darkmoon233, 2019-07-28:
Thank you very much, the examples you provided have helped me understand this problem, which is exactly what I want.
Further, if I want to use WASM to call C++ functions, can the wasm-micro-runtime provide an API? Or should I implement the pointer transfer through the location of the memory?

darkmoon233, 2019-07-29:
I noticed that JavaScript can load some parameters (such as some externally defined functions) into the instantiation process through the `WebAssembly.instantiate()` .
`WebAssembly.instantiate()` can be used like that.
```JavaScript
const importObject = {
  imports: {
    imported_func(arg) {
      console.log(arg);
    }
  }
};
async function f(){
    const response = await fetch('test.wasm');
    const bufferSource = await response.arrayBuffer();

    const result = await WebAssembly.instantiate(bufferSource, importObject);
    const {instance:{exports:exported_func}} = result;

    exported_func();
}
```

`wasm_runtime_instantiate()` can create a wasm instance in wasm-micro-runtime. Can it pass the parameters that need to be loaded in a similar way to `WebAssembly.instantiate()`?

wenyongh, 2019-07-29:
Hi, @darkmoon233, iwasm supports "externally defined functions" by registering the native functions into the array of extended_native_symbol_defs, which must be provided by developer. In iwasm_cpp.zip, it is located in src/ext_lib_export.cpp.
And also you can ref to the section "The mechanism of exporting native API to WASM application" in wamr README for a help.
I uploaded a new attachment to show how to call a native extended func:
in src/ext_lib_export.cpp, a function void test() is implemented to call the cpp class Test's member function. And function test() is registered in array extended_native_symbol_defs.
And in wasm app, it calls the test() function in main() function.

[iwasm_cpp.zip](https://github.com/intel/wasm-micro-runtime/files/3440766/iwasm_cpp.zip)

darkmoon233, 2019-07-31:
I can now use micro-wasm-runtime for the most basic cross-call of C++ and wasm! Your example is very clear and helpful!

closed 2019-10-08 by wenyongh