wasm-micro-runtime/wasm-micro-runtime #92: Does micro-wasm-runtime have a way to call external pointers directly?

darkmoon233, 2019-08-08:
I have noticed that:
>Security attention: A WebAssembly application should only have access to its own memory space. As a result, the integrator should carefully design the native function to ensure that the memory accesses are safe. The native API to be exported to the WASM application must:  
>- Only use 32 bits number for parameters
>- Should not pass data to the structure pointer (do data serialization instead)
>- Should do the pointer address conversion in the native API
>- Should not pass function pointer as callback

But if i can't access the memory outside ,the wasm can do little things.

I want to implement cross-calling of wasm and c++，so i have some questions about pointer!
1. I want use wasm to access the memory out of wasm world which belongs to the C++
process which creates the wasm instance by pointer.
But when i pass a extternal pointer to wasm, it occurs some errors.Does the micro-wasm-runtime refuse access the external memory?Or i did something wrong? (Attachment is the complete code 
 if you need[pointer-test.zip](https://github.com/intel/wasm-micro-runtime/files/3480986/pointer-test.zip)）

```cpp
// cpp code
    struct cppStruct
    {
        char c;
        short s;
    };
cppStruct stu ={'a',0};
cppStruct* _stu_ptr = &stu;
int app_add = wasm_runtime_addr_native_to_app(wasm_module_inst,(void*)_stu_ptr);
func_argv[0] = app_add;
//call wasm funcion
wasm_func = wasm_runtime_lookup_function(wasm_module_inst, "modify_cpp_struct", "(i32)");
wasm_runtime_call_wasm(wasm_module_inst, NULL, wasm_func, 1, func_argv);
```

```cpp
// wasm-src code
struct cppStruct{
        char c;
        short s;
 };
void modify_cpp_struct(cppStruct* offset){
	wasm_module_inst_t module_inst = wasm_runtime_get_current_module_inst();
	void* temp_ptr  = (void*)offset;
	size_t temp_add = (size_t)temp_ptr;

       // the following 4 function will occer error ‘Segmentation fault (core dumped)’ so i have no way to use the pointer to access the memory in the c++ precess

	// wasm_runtime_validate_app_addr(module_inst,temp_add,1);
	// wasm_runtime_validate_native_addr(module_inst,(void*)offset,1);
	// wasm_runtime_addr_native_to_app(module_inst,(void*)offset);
	// wasm_runtime_addr_app_to_native(module_inst,(int)temp_add);
} 
```
2. The function `wasm_runtime_addr_native_to_app(wasm_module_inst,(void*)_stu_ptr)` return int32, if i use clang or emcc compile the C++ file by 64-bit,the address of wasm is 8byte.In that case,does the function can work currently?
3. The g++ can't find the library `iwasm` when i use g++ with param -m64 to compile the cpp file to iwasm(not the wasm-src).Dose the library only can be used in 32-bit?

wenyongh, 2019-08-08:
Hi, WASM app’s memory space is different from native memory space, the former is offset of relative address, the latter is absolute address. WASM app should only have access to its own memory space, here iwasm also initializes an private app heap, which allows wasm app to access it besides with its linear memory space. The app heap size is passed as parameter to wasm_runtime_instantiate function: 
wasm_runtime_instantiate(wasm_module,
                                                      32 * 1024, /* stack size */
                                                      32 * 1024,  /* heap size */
And VM provides API to allocate memory from the app heap (see wasm_export.h):
int32_t
wasm_runtime_module_malloc(wasm_module_inst_t module_inst, uint32_t size);

So for question 1, if we don't consider the structure layout issue, you may allocate memory from app heap, and pass it to function of wasm app, the source code may be like:
  int app_addr = wasm_runtime_module_malloc(wasm_module_inst, sizeof(cppStruct));
  if (app_addr == 0) { handle error; }
  cppStruct* stu_ptr = wasm_runtime_addr_app_to_native(wasm_module_inst, app_addr);
  stu_ptr->c = 'a';
  stu_ptr->s = 0;
  func_argv[0] = app_addr;
  ...

The structure layout may be different between native and WASM app, so we suggest to do data serialization between them, especially when there are pointer fields in structure. 

And please don't call wasm_runtime_xxx API's in wasm app's source code..

For question 2, we currently only support wasm32.

For question 3, libiwasm.so can be built in both 32-bit mode and 64-bit mode, in file iwasm/products/linux/CMakeLists.txt, set the flag BUILD_AS_64BIT_SUPPORT to "NO" or "YES":
  set (BUILD_AS_64BIT_SUPPORT "NO")  => 32 bit
  set (BUILD_AS_64BIT_SUPPORT "YES")  => 64 bit

darkmoon233, 2019-08-09:
The heap memory allocated to wasm is of course a way of communication between C++ and wasm. But this makes me feel not direct. Using pointers is too important for C++.

Because the memory space of wasm should be part of the memory space of the c++ program that loads it (my guess, is this correct?).

If this is the case, what limits this access?

wenyongh, 2019-08-09:
For security reason, WebAssembly's "code is validated and executes in a memory-safe, sandboxed environment preventing data corruption or security breaches."

wasm app's address space and native address space are different concept, wasm vm strictly checks the app's address to ensure that it is in the range of app's memory space.

If the pointer is generated by native lib and is out of the range of wasm app's memory space, we had better not access the pointer directly in wasm app. We can implement some native lib API's to access the pointer, and let wasm app calls these API's instead of accessing the pointer directly.

darkmoon233, 2019-08-14:
Accessing addresses out of the WASM's memory  is not possible. There is an external space address that WASM cannot be accessed because the address range of the wasm process is smaller than the corresponding C++ process.

Imagine this，If we create a wasm instance, set the wasm address space to be the same as the C++ space (of course, address space security should be considered), and the starting address of wasm is no longer 0, but starts at a specified location. In this case, all wasm addresses can correspond to C++ addresses. All C++ addresses wasm can be accessed (by conversion) by this method.
Do you think this assumption is feasible?

wenyongh, 2019-08-15:
The wasm address space size is specified by toolchain with page count, each page is 64K, and VM initialize the memory by allocating memory of suitable size from system, so it is difficult to set the wasm address space to be the same as the C++ space.

And also the toolchain generates wasm load/store bytecodes with offset operand whose range is from 0 to wasm_mem_space_size to access the wasm memory. So for wasm vm, the address it gets from bytecode is relative offset, and it adds the offset with wasm memory base address to access the  wasm memory. The wasm memory base address is absolute address.

If you insist on directly accessing external C++ addresses in wasm app, you can try passing offset = c++_addr - wasm_mem_base_addr to wasm app, and removing the boundary check in wasm interpreter. But there may be other potential issues unexpected.

darkmoon233, 2019-08-16:
Your suggestion is very meaningful, i'm having a try.
Consider another way, we provide a memory space to the wasm, if the size of the space is appropriate, we can also make wasm access to the C++ address (of course need for address translation). 
I know that JavaScript can provide that method, but whether it is provided in wasm-micro-runtime. I tried the `wasm_runtime_load` method, but this method does not have the ability to be associated with wasm memory.

This is the way JavaScript provide external memory to wasm.
```JavaScript
const memory = new WebAssembly.Memory({initial: 20});
const imports = {
  env: {
    memory: memory
  }
};
```

wenyongh, 2019-08-21:
Hi, JavaScript can provide an external memory to wasm, but if the memory is enlarged with opcode memory.grow, the memory space may be changed. And also it cannot resolve the address conversion issue.

Actually your requirements include: (1) wasm app is able to access the native memory space; (2) the memory space can be another external memory space besides wasm app's linear memory space and app's heap space.

For (1), we have provided API's to access wasm app memory space in native, but we haven't provided the similar API's for wasm app, like, wasm app gets native absolute address and knows its data layout, then it wants to access its data. So we implemented these API's for wasm app, including:

    uint64_t wasm_runtime_get_current_module_inst();
    bool wasm_runtime_validate_native_addr(uint64_t module_inst, uint64_t native_ptr, uint32_t size);
    bool wasm_runtime_validate_app_addr(uint64_t module_inst, int32_t app_offset, uint32_t size);
    int32_t wasm_runtime_addr_native_to_app(uint64_t module_inst, uint64_t native_ptr);
    uint64_t wasm_runtime_addr_app_to_native(uint64_t module_inst, int32_t app_offset);

Note that the type of module instance and native pointer is uint64_t since native pointer may be 64-bit, but wasm app's pointer is only 32-bit, data loss may occur if we define them as void*,  which is compiled to type i32 in wasm.

So if you get a native absolute address in wasm app, you can call wasm_runtime_get_current_module_inst() and wasm_runtime_addr_native_to_app() to convert the native address to wasm app's int32 offset, and then convert the i32 offset to a pointer type, like char*, to access the memory.

For (2), we provide an API wasm_runtime_set_ext_memory() to set the external memory space to module instance after it is instantiated. The API may introduce security issue, so we disabled it with macro WASM_ENABLE_EXT_MEMORY_SPACE by default, you can set it to 1 in file core/shared-lib/include/config.h or add_definitions("-DWASM_ENABLE_EXT_MEMORY_SPACE") in CMakeLists.txt to enable it.

The patch has been merged, and here is an example to use these API's, hope it is helpful for you.
[iwasm_cpp.zip](https://github.com/intel/wasm-micro-runtime/files/3524514/iwasm_cpp.zip)

In native ext_lib_export.c, it creates a string in ext memory space and pass the native pointer to wasm app; and in wasm app, it converts the native pointer to offset and then access it.

wenyongh, 2019-08-21:
Currently we map wasm memory space address to 0 - 1G for wasm app, and map app heap space address to 1G - 2G. And for external memory space, we map it to 2G-3G. So if you print the pointer address of external memory, e.g. its start address with "%p", the output is: 0x80000000, (=2G).

closed 2019-10-08 by wenyongh