wasm-micro-runtime/wasm-micro-runtime #100: Is it possible to use wasm-micro-runtime as scripting engine?

blockspacer, 2019-08-16:
**Use case**: Replace scripting engines like Lua with a faster, light-weight system that’s programmable in several languages (wasm-micro-runtime). 

**Question**: Is it possible to make wasm <-> C++ binding easier? (for example, disable some security features e.t.c.). 

**Issue**: It is hard import an external class e.t.c. https://github.com/intel/wasm-micro-runtime/issues/95
> for the security and toolchain reason, it is really not easy to directly access the external class/structure created outside wasm app.

wenyongh, 2019-08-21:
Hi, please see [#92](https://github.com/intel/wasm-micro-runtime/issues/92), we have implemented address validation and conversion API's for both native and app. So now wasm app should be able to convert native pointer which is absolute address to app's address which is relative offset, and then access it, please see the sample. These API's include:

    wasm_runtime_get_current_module_inst();
    wasm_runtime_validate_native_addr();
    wasm_runtime_validate_app_addr();
    wasm_runtime_addr_native_to_app();
    wasm_runtime_addr_app_to_native();

blockspacer, 2019-08-21:
Awesome work. Thank you for that great Project!

closed 2019-10-08 by wenyongh