wasm-micro-runtime/wasm-micro-runtime #101: Can be used on nrf51

aibeb, 2019-08-20:
Can be used on nrf51 with zephyr?

wenyongh, 2019-08-22:
Hi, @aibeb, we haven't tested wamr on nrf51, but we have enabled wamr on stm32 board with Zephyr, which may be similar to your requirements. The samples can be found in: 
[core/iwasm/products/zephyr/simple/](https://github.com/intel/wasm-micro-runtime/tree/master/core/iwasm/products/zephyr/simple)
which is a standalone wasm vm to run a simple wasm app

[samples/gui/wasm-runtime-wgl/](https://github.com/intel/wasm-micro-runtime/tree/master/samples/gui/wasm-runtime-wgl)
which builds wasm vm, app manager and littlevgl lib together to run wasm app developed with ui APIs wrapperred by ourselves, most of the littlevgl lib code is built into wamr but not wasm app;

and [samples/littlevgl/vgl-wasm-runtime/](https://github.com/intel/wasm-micro-runtime/tree/master/samples/littlevgl/vgl-wasm-runtime)
which builds wasm vm and app manager together to run wasm app developed with littlevgl lib, most of the littlevgl lib code is built into the wasm app but not wamr.

aibeb, 2019-08-22:
@wenyongh 

I used "core/iwasm/products/zephyr/simple/" to do the test, it seems that the memory does not meet the demand.

`[ 96%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       80828 B       256 KB     30.83%
            SRAM:      535608 B        32 KB    1634.55%
        IDT_LIST:          56 B         2 KB      2.73%`

wenyongh, 2019-08-22:
Your total memory size is 32KB? It may be too small to run the wasm app. But you can have a try:
(1) Decrease the global heap size:
    in main.c, modify global_heap_buf size from 512KB to 20KB;

(2) Decrease the app heap size and stack size:
    in main.c, modify
    wasm_runtime_instantiate(.., 8 * 1024, 8 * 1024, ..) to
    wasm_runtime_instantiate(.., 4 * 1024, 2 * 1024, ..)
    The stack is for interpreter to store operands, if it is too small, stack overflow exception may be thrown;
    The app heap is to allocate memory for wasm app.

(3) Decrease the default main thread size:
    in main.c, modify DEFAULT_THREAD_STACKSIZE from 6K to 3K

(4) Decrease the memory page size of wasm app, by default specified by WASM spec, each memory page size is 64K, but you can decrease it to 8K to have a try:
    in core/iwasm/runtime/vmcore-wasm/wasm.h
Note that if the wasm app's footprint is large enough, this may cause potential issues.

vshymanskyy, 2020-01-12:
@aibeb it seems that your nRF51 has 32Kb RAM.
This is good news as there are 16Kb variants, and those would be very limited.
[Wasm3](https://github.com/wasm3/wasm3) engine can run even on 16Kb version, but it's much more practical with 32Kb.
There's a [PlatformIO example](https://github.com/wasm3/wasm3/blob/94ef35d84318d5f0793eb00b62d7125812be98b3/platforms/arduino/platformio.ini#L58-L76), but you should be able to integrate into your build environment without issues.

closed 2020-09-03 by wenyongh