wasm-micro-runtime/wasm-micro-runtime #112: Buildroot support

beriberikix, 2019-09-07:
I've been playing around with [buildroot](https://buildroot.org/), a system that makes it easy to build custom Linux images for embedded devices like Raspberry Pi. I was able to create an image for a Pi 3 with little effort that boots in 4 seconds.

I'm investigating how to make WAMR a [userspace package](https://buildroot.org/downloads/manual/manual.html#adding-packages) and it doesn't seem too hard.

Has anyone else considered this? Anyone with buildroot experience who might be able to help?

wenyongh, 2019-09-09:
Hi @beriberikix, we have no experiences on buildroot, it seems to be complex to configure the buildroot scripts and automatically build wamr into buildroot. But the buildroot generates rootfs package and the gcc toolchain, could you try using its toolchain to compile wamr into a executable file, and put it under rootfs package to run?

beriberikix, 2019-09-10:
I spent some time this weekend experimenting with buildroot's packaging system with little success. I posted in their forum asking for help: http://lists.busybox.net/pipermail/buildroot/2019-September/259086.html

zerotri, 2019-10-05:
In case it is useful to anyone else in the future, I've implemented this in a minimal buildroot package here: https://github.com/zerotri/zerotwo/commit/5a8cd3e94e517a3f2bb6ddbef0f0b935fbe32a07#diff-07db48543851da2e2eec161847f57a81

This doesn't do anything fancy with the builds, if there's a specific feature that should be added please let me know. My experience with web assembly is pretty much limited to getting this up and running in buildroot.

Tested on aarch64(Raspberry Pi 4 + Qemu) and arm32v7 (Raspberry Pi 4) with a simple example from the old readme.

beriberikix, 2019-12-08:
@zerotri thank you sooo much! I feel terrible not replying since I was the one who asked for your help 🙁. But I'm picking it up again! n00b question - I don't see the WAMR package in the menuconfig. How do I source it?

zerotri, 2019-12-10:
You would need to tell buildroot where the package info is via the Config.in and the appropriate .mk in the `package/` directory of buildroot. I do this via a buildroot external tree.
You can grab my external tree setup here: 
https://github.com/zerotri/zerotwo
You can cd into the repo and run `git submodule update --init --recursive` to get the buildroot sources, then you can simply run `make raspberrypi4_defconfig` as you normally would to get started. Then you should see the package in `make nconfig` under the very bottom menu.

closed 2020-09-20 by wenyongh