wasm-micro-runtime/wasm-micro-runtime #113: Error when exporting native api on zephyr

amalabey, 2019-09-08:
Hi,

I am trying to export timer apis on zephyr (board=esp32) and get below error when compiling. What am I missing? Do you have any concrete examples of exposing apis on zephyr?

*ext_lib_export.c" in my zephyr simple example:
```
#include "lib_export.h"
#include "wasm_app.h"

static NativeSymbol extended_native_symbol_defs[] = {
    EXPORT_WASM_API(wasm_create_timer),
    EXPORT_WASM_API(wasm_timer_set_interval),
    EXPORT_WASM_API(wasm_timer_cancel),
    EXPORT_WASM_API(wasm_timer_restart)
};

#include "ext_lib_export.h"
```

* CmakeList.txt included directories
```
include_directories (${IWASM_ROOT}/runtime/include
                     ${IWASM_ROOT}/runtime/platform/include
                     ${IWASM_ROOT}/runtime/platform/zephyr
                     ${IWASM_ROOT}/runtime/vmcore-wasm
                     ${SHARED_LIB_ROOT}/include
                     ${SHARED_LIB_ROOT}/platform/include
                     ${SHARED_LIB_ROOT}/platform/zephyr
                     ${IWASM_ROOT}/lib/native/extension/connection/zephyr
                     ${IWASM_ROOT}/lib/app-libs/base
                     ${IWASM_ROOT}/extension/sensor
                     ${IWASM_ROOT}/lib/native-interface
                     ${IWASM_ROOT}/lib/app-libs/extension/connection
                     ${IWASM_ROOT}/lib/app-libs/extension/gui
                     )
```

*output from ninja below

```
[108/115] Building C object CMakeFiles/app.dir/src/ext_lib_export.c.obj
FAILED: CMakeFiles/app.dir/src/ext_lib_export.c.obj 
ccache /home/amal/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DBUILD_VERSION=v2.0.0-rc3-13-g0a6a2159bd08 -DKERNEL -DNVALGRIND -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../iwasm/runtime/include -I../iwasm/runtime/platform/include -I../iwasm/runtime/platform/zephyr -I../iwasm/runtime/vmcore-wasm -I../iwasm/../shared-lib/include -I../iwasm/../shared-lib/platform/include -I../iwasm/../shared-lib/platform/zephyr -I../iwasm/lib/native/extension/connection/zephyr -I../iwasm/lib/app-libs/base -I../iwasm/extension/sensor -I../iwasm/lib/native-interface -I../iwasm/lib/app-libs/extension/connection -I../iwasm/lib/app-libs/extension/gui -I/home/amal/work/iot/esp32-zephyr/zephyr/kernel/include -I/home/amal/work/iot/esp32-zephyr/zephyr/arch/xtensa/include -I/home/amal/work/iot/esp32-zephyr/zephyr/include -I/home/amal/work/iot/esp32-zephyr/zephyr/include/drivers -Izephyr/include/generated -I/home/amal/work/iot/esp32-zephyr/zephyr/soc/xtensa/esp32 -I/home/amal/work/iot/esp32-zephyr/zephyr/soc/xtensa/esp32/include -I/home/amal/work/iot/esp32-zephyr/modules/hal/esp-idf/zephyr/../components/esp32/include -I/home/amal/work/iot/esp32-zephyr/modules/hal/esp-idf/zephyr/../components/soc/esp32/include -isystem /home/amal/work/iot/esp32-zephyr/zephyr/lib/libc/minimal/include -isystem /home/amal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/include -isystem /home/amal/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/include-fixed -Os -imacros/home/amal/work/iot/esp32-zephyr/zephyr/samples/simple/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacros/home/amal/work/iot/esp32-zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -mlongcalls -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/ext_lib_export.c.obj -MF CMakeFiles/app.dir/src/ext_lib_export.c.obj.d -o CMakeFiles/app.dir/src/ext_lib_export.c.obj   -c ../src/ext_lib_export.c
In file included from /home/amal/work/iot/esp32-zephyr/zephyr/include/net/net_ip.h:27:0,
                 from /home/amal/work/iot/esp32-zephyr/zephyr/include/net/net_pkt.h:26,
                 from ../iwasm/../shared-lib/platform/zephyr/bh_platform.h:37,
                 from ../iwasm/lib/native-interface/native_interface.h:22,
                 from ../iwasm/lib/app-libs/base/wasm_app.h:35,
                 from ../src/ext_lib_export.c:18:
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:20:33: error: expected declaration specifiers or '...' before '(' token
 #define __bswap_32(x) ((u32_t) ((((x) >> 24) & 0xff) | \
                                 ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:107:30: note: in expansion of macro '__bswap_32'
 #define sys_cpu_to_be32(val) __bswap_32(val)
                              ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/net/net_ip.h:121:18: note: in expansion of macro 'sys_cpu_to_be32'
 #define htonl(x) sys_cpu_to_be32(x)
                  ^
../iwasm/lib/app-libs/base/bh_platform.h:51:8: note: in expansion of macro 'htonl'
 uint32 htonl(uint32 value);
        ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:20:33: error: expected declaration specifiers or '...' before '(' token
 #define __bswap_32(x) ((u32_t) ((((x) >> 24) & 0xff) | \
                                 ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:106:30: note: in expansion of macro '__bswap_32'
 #define sys_be32_to_cpu(val) __bswap_32(val)
                              ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/net/net_ip.h:97:18: note: in expansion of macro 'sys_be32_to_cpu'
 #define ntohl(x) sys_be32_to_cpu(x)
                  ^
../iwasm/lib/app-libs/base/bh_platform.h:52:8: note: in expansion of macro 'ntohl'
 uint32 ntohl(uint32 value);
        ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:19:33: error: expected declaration specifiers or '...' before '(' token
 #define __bswap_16(x) ((u16_t) ((((x) >> 8) & 0xff) | (((x) & 0xff) << 8)))
                                 ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:101:30: note: in expansion of macro '__bswap_16'
 #define sys_cpu_to_be16(val) __bswap_16(val)
                              ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/net/net_ip.h:113:18: note: in expansion of macro 'sys_cpu_to_be16'
 #define htons(x) sys_cpu_to_be16(x)
                  ^
../iwasm/lib/app-libs/base/bh_platform.h:53:8: note: in expansion of macro 'htons'
 uint16 htons(uint16 value);
        ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:19:33: error: expected declaration specifiers or '...' before '(' token
 #define __bswap_16(x) ((u16_t) ((((x) >> 8) & 0xff) | (((x) & 0xff) << 8)))
                                 ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/sys/byteorder.h:100:30: note: in expansion of macro '__bswap_16'
 #define sys_be16_to_cpu(val) __bswap_16(val)
                              ^
/home/amal/work/iot/esp32-zephyr/zephyr/include/net/net_ip.h:89:18: note: in expansion of macro 'sys_be16_to_cpu'
 #define ntohs(x) sys_be16_to_cpu(x)
                  ^
../iwasm/lib/app-libs/base/bh_platform.h:54:8: note: in expansion of macro 'ntohs'
 uint16 ntohs(uint16 value);
        ^
In file included from ../src/ext_lib_export.c:17:0:
../src/ext_lib_export.c:22:21: error: 'wasm_timer_set_interval' undeclared here (not in a function)
     EXPORT_WASM_API(wasm_timer_set_interval),
                     ^
../iwasm/runtime/include/lib_export.h:29:44: note: in definition of macro 'EXPORT_WASM_API'
 #define EXPORT_WASM_API(symbol)  {#symbol, symbol}
                                            ^
ninja: build stopped: subcommand failed.
```

wenyongh, 2019-09-09:
Hi @amalabey, thanks for using wamr, please don't include wasm_app.h in ext_lib_export.c, it may cause compiler error.

Not sure what is your requirements of porting wamr to Zephyr on esp32? Do you want to use app manager to download the wasm app? Currently the timer API needs support of app manager, there are some samples to ref:
(1) samples/simple, sample to start app manager with iwasm vm, and download wasm app e.g. timer app from host to run; currently it only runs on Linux, we may port it to Zephyr soon, and maybe you can have a try;
(2) samples/littlevgl/vgl-wasm-runtime/zephyr-build: run wasm app developed with littlevgl graphics library, we build littlevgl lib into wasm app;
(3) samples/gui/wasm-runtime-wgl/zephyr-build: run wasm app developed with wamr graphics library, which is wrapped based on littlevgl lib; we build littlevgl lib into wamr but not wasm app;

amalabey, 2019-09-13:
Thanks, I managed to get it working on Zephyr (on ESP32). This was just for fun as a proof-of-concept. Great work!

At the moment, I just include the generated wasm in test_wasm.h and flash the re-compiled sample onto esp32.

closed 2019-09-13 by amalabey