wasm-micro-runtime/wasm-micro-runtime #121: How to port to a different RTOS?

amalabey, 2019-09-14:
Can you provide some guidance/documentation on how to enable support to another RTOS (e.g. FreeRTOS)? I got the runtime working with Zephyr on ESP32, but unfortunately, I can't get the WIFI working because ESP-IDF uses FreeRTOS.

wenyongh, 2019-09-15:
Hi, suppose the platform directory name is freertos, there are three folders to be created:
(1) core/shared/platform/freertos
   files: bh_platform.h/c, bh_assert.c, bh_definition.c, bh_platform_log.c,
           bh_time.c bh_thread.c,
           bh_match.c(if some math functions are not supported by FreeRTOS)
(2) core/iwasm/runtime/platform/freertos
    file: wasm_native.c with function wasm_platform_native_func_lookup(),
          which normally just returns NULL.
(3) core/iwasm/products/freertos
    files: main.c, ext_lib_export.c and makefile e.g. CMakeLists.txt

Could you ref to the implementation of other platforms e.g. Linux and Zephyr?

amalabey, 2019-09-15:
Thanks, appreciate the quick response!

closed 2019-09-15 by amalabey

vshymanskyy, 2020-01-06:
@amalabey did you get the wifi working?

amalabey, 2020-01-06:
Sorry, no it turned out to be too much work to support FreeRtos and couldn’t complete.

vshymanskyy, 2020-01-06:
@amalabey you might be interested in this https://github.com/bytecodealliance/wasm-micro-runtime/issues/134#issuecomment-571178221