wasm-micro-runtime/wasm-micro-runtime #123: WAMR relicensing coordination

joshtriplett, 2019-09-17:
WAMR is currently licensed under the Apache 2.0 license. To better collaborate with other projects like wasmtime, cranelift, lucet, and LLVM, Intel would like to match the license of those projects, which use the [Apache 2.0 license with LLVM exception](https://llvm.org/docs/DeveloperPolicy.html#new-llvm-project-license-framework). The LLVM exception keeps all the properties of the Apache 2.0 license, and improves license compatibility.

In order to make this change, we need your help; we're seeking permission from the non-Intel contributors to WAMR to make this change. The following people have made contributions to WAMR:

- [ ] @zhujinzhou
- [ ] @yiting16
- [ ] @beriberikix
- [ ] @jondong
- [ ] @0xflotus
- [ ] @hakilebara
- [ ] @pcmill
- [ ] @offercomes

If this seems reasonable to you, could you please respond to this issue with a comment? (Just saying "sure" is fine; we'll coordinate the list.)

Once we have approval from the folks listed above, I'll post a PR adding the LLVM exception text to the license.

Thank you again for your contributions to WAMR!

beriberikix, 2019-09-17:
Sure! Glad to see the collaboration 😁

0xflotus, 2019-09-17:
Sure :)

yiting16, 2019-09-18:
Sure.

jondong, 2019-09-18:
Sure! nice to see the collaboration

zhujinzhou, 2019-09-18:
sure!!!

pcmill, 2019-09-18:
Sure, I just made some changes to the README :smiley:

joshtriplett, 2019-11-11:
Fixed in #137.

closed 2019-11-11 by joshtriplett