wasm-micro-runtime/wasm-micro-runtime #132: WebAssembly C and C++ API compatibility

beriberikix, 2019-10-17:
https://github.com/WebAssembly/wasm-c-api

The We assembly wg has been working on a standard c/c++ API for runtimes to increase compatibility. V8 is the current reference implementation and it would be awesome if WAMR followed. For example it would allow constrained applications that originally use V8 for their prototype to switch to WAMR when they move to the constrained domain.

wenyongh, 2019-10-22:
Hi, @beriberikix, do you know what is the difference between wasm-c-api and wasi? It seems that wasi also supports most of the standard c/c++ API's, and is accepted by some wasm vms. And what is the progress of wasm-c-api and wasm-c-api in v8? The website says "No docs yet".

beriberikix, 2019-10-22:
@rossberg @sunfishcode @sbc100

The basic documentation ifor V8 can be seen [here](https://docs.google.com/document/d/1oFPHyNb_eXg6NzrE6xJDNPdJrHMZvx0LqsD6wpbd9vY/edit, known as Wee8.

sbc100, 2019-10-22:
WASI is standard set of API for use by webassembly modules, whereas the wasm-c-api is a standard for embedded WebAssembly VMs in other programs. 

Another way of putting it: WASI is for developers authoring wasm modules, and the wasm-c-api is for developers who want to embed wasm in existing C/C++ applications.

wenyongh, 2019-10-22:
Thanks a lot for the explanation, @sbc100. So wasm-c-api is a set of API exposed by WebAssembly VM for other program to interact with it,  like instantiate a wasm module, lookup/call wasm function, get/set wasm global and so on, similar with Java JNI of JVM. We may spend some time evaluating it.

beriberikix, 2019-10-22:
I believe the wasm-c-api would compliment (or possible replace?) the current set of [embed APIs](https://github.com/intel/wasm-micro-runtime/blob/master/doc/embed_wamr.md).

wenyongh, 2019-10-23:
Yes, WAMR [embed APIs](https://github.com/intel/wasm-micro-runtime/blob/master/doc/embed_wamr.md) (exported in runtime/include/wasm_export.h) seems to be subset of the wasm-c-api, but WAMR provides some APIs that wasm-c-api doesn't have, e.g. the address conversion between native absolute address and wasm app relative address.

From its [C header file](https://github.com/WebAssembly/wasm-c-api/blob/master/include/wasm.h), the wasm-c-api, wasm-c-api provides more APIs to access the internal data of WASM module, e.g. function type, function, table, global, memory, like reflection feature of Java.

rossberg, 2019-10-23:
Drive-by comment: functions, tables, globals, memories are not just internals of Wasm modules. It's an essential part of the Wasm semantics that modules can import these things, and not just from other modules but also from the host. Consequently, the API needs to provide a way to construct these things (cf. the JavaScript API as well).

wenyongh, 2019-10-23:
Yes, @rossberg, thanks for your correction.

wenyongh, 2020-09-01:
Most APIs of wasm-c-api were implemented in WAMR, seems that we can close the issue.

closed 2020-09-01 by wenyongh