wasm-micro-runtime/wasm-micro-runtime #141: How does Request/Response & Pub/Sub work?

asim, 2019-11-12:
Hi, can you please explain in detail how and what the request/response and pub/sub interfaces connect to in the underlying host infrastructure? The wasm runtime appears to provide api interfaces for this form of communication but doesn't specify what that connects to exactly to facilitate the communication.

xwang98, 2019-11-13:
@asim Thanks for your interest. 

The communication to remote applications can be over any kind of physical link such as Socket, Serial port, SPI, BLE etc. Supposing your software will embed the WAMR, so when your software received any message from the link, just call the function aee_host_msg_callback() to handle the message.  And also your software need to provide WAMR your function for transmitting a message to remote by calling app_manager_host_init().

The host-tool under test-tools folder is a sample as remote application communicating with the WASM apps.

asim, 2019-11-13:
Thank you for the response. Considering the plug and play nature of [go-micro](https://github.com/micro/go-micro), our pubsub/request-response interfaces and our [micro](https://github.com/micro/micro) runtime 😉. We think this would be a great fit for wasm based microservices. Hopefully we'll be able to contribute something soon.

xwang98, 2019-11-13:
That would be awesome! @asim

closed 2019-11-27 by xwang98