wasm-micro-runtime/wasm-micro-runtime #152: ARM build with west

Papadiddypub, 2019-12-27:
Hi, following the example to build for zephyr with west (west build -p auto -b qemu_cortex_m3) the build fails with an error (/iwasm/runtime/vmcore-wasm/invokeNative_ia32.s:10: Error: unrecognized symbol type ""). The selected biuld target is: Build as target X86_32
if I build for x86 (west build -p auto -b qemu_x86 the build works fine. 
Using ninja the same error occurs.

wenyongh, 2019-12-27:
Hi, the board qemu_cortex_m3 should be THUMB arch, could you edit file core/iwasm/products/zephyr/simple/CMakeLists.txt, modify set (BUILD_TARGET "X86_32") to set (BUILD_TARGET "THUMB"), and try again?

Papadiddypub, 2019-12-27:
Hi,
I changed the BUILD_TARGET in the file core/iwasm/products/zephyr/simple/CMakeLists.txt without success (no changes in the build process). Afterwards I changed the BUILD_TARGET in zephyrprojects/zephyr/sample/simple/ CMakeLists.txt (setup according to project description). Now the BUILD_TARGET changed to THUMB. The build failed during linking with the following message: zephyr/zephyr_prebuilt.elf section `bss' will not fit in region `SRAM'

Afterwards I build for stm32 (-b nucleo_f746zg) and the build was successful.

Papadiddypub, 2019-12-30:
Thanks for the quick help. Issue is closed. Usage of BUILD_TARGET "THUMB" in CMakeList.txt in build folder resolved the build problem.

closed 2019-12-30 by Papadiddypub