wasm-micro-runtime/wasm-micro-runtime #159: Redeclaration and conflicting types in wasm_export.h and wasm_runtime.h.

poemm, 2020-01-23:
When the following are included, the compiler complains of redeclaration and conflicting types.

core/iwasm/include/wasm_export.h
core/iwasm/interpreter/wasm_runtime.h

Perhaps duplicated parts should be put into core/iwasm/common/. But I prefer you to advise on how to fix this in a way which aligns with your design goals.

wenyongh, 2020-01-23:
Hi, could you try just including wasm_export.h, and not including wasm_runtime.h? In our design, wasm_export.h is exposed to developer to call the wasm vm api, and wasm_runtime.h in only used by interpreter internally.

poemm, 2020-01-23:
Thank you for helping me. Correct, I can include just one or the other header, but I need both. 

The Wasm spec says "A host function may also modify the store." I don't need the whole store, just the calling module's memory. Specifically, I want to do the following inside the host function.

```
WASMMemoryInstance *mem = ((const WASMModuleInstance*)exec_env->module_inst)->default_memory;
```

Please give advice on a nice way to do this.

wenyongh, 2020-01-24:
There may be two methods to resolve the issue: one is to add macro to control the structure definition in each redeclatation structure in the two header files, for example:
#ifndef _defined_structure_xxx
define the structure
#define _defined_structure_xxx
#endif

the second is to include wasm_export.h only, and add related APIs in it to access the data structure you want, e.g.
void *wasm_runtime_get_mem_addr(module_inst);
uint32_t wasm_runtime_get_mem_size(module_inst);
Note that mem base addr may be changed after memory.grow opcode is executed.

poemm, 2020-01-24:
Thank you. Those two options work, but require patching `wasm_export.h`. To avoid changing your library, I copyied the definitions of `WASMModuleInstance` and `WASMMemoryInstance` into my program. And added the following dummy definitions.

```
typedef struct WASMTableInstance{int a;} WASMTableInstance;
typedef struct WASMGlobalInstance{int a;} WASMGlobalInstance;
typedef struct WASMFunctionInstance{int a;} WASMFunctionInstance;
typedef struct WASMExportFuncInstance{int a;} WASMExportFuncInstance;
typedef struct WASMModule{int a;} WASMModule;
typedef struct WASMExecEnv{int a;} WASMExecEnv;
```

Now everything compiles and works as I want. But I would rather permanently patch `wasm_runtime.h`. Would you object to having `wasm_runtime_get_mem_addr` and `wasm_runtime_get_mem_size` in `wasm_export.h`?

> Note that mem base addr may be changed after memory.grow opcode is executed.

Thank you for this important tip. The host function gets the most recent mem base addr, then `memcpy` to/from the memory. I must make sure that there is no chance of the mem base addr changing while I am in the host function.

wenyongh, 2020-01-24:
Glad to hear that. It may be good to add the two apis in wasm_export.h, but we need to discuss it internally. Another issue is that have you tested the aot and will you use aot? For aot, the related data structure is AOTModuleInstance, and you need to implement the two apis of aot version. For example, in wasm_export.h, define the two apis, and in wasm_runtime_common.c, implement the two apis by checking the module type of module_inst, and call the related apis of version interpreter or aot accordingly: in interpreter, implement wasm_get_mem_addr and wasm_get_mem_size, in aot, implement aot_get_mem_addr and aot_get_mem_size. Please ref to other apis in wasm_runtime_common.c.

wenyongh, 2020-03-27:
The header files were re-orged, developer can include "wasm_export.h" to call vmcore APIs, or include "wasm_runtime_common.h" to use the vmcore internal data structures, or include both of them if needed. So close this issue.

closed 2020-03-27 by wenyongh