wasm-micro-runtime/wasm-micro-runtime #160: Cannot run AoT-compiled ELF file in iwasm runtime

masomel, 2020-01-27:
I compiled a WASM bytecode file using `wamrc` with the `--format=object` flag. I can run the WASM _bytecode_ file directly in `iwasm`  (under product-mini/platforms/linux) successfully, but when I try to run the AoT-compiled object file in `iwasm`, I get the following error:

"WASM module load failed: magic header not detected".

I am running on Ubuntu 18.04.1.

wenyongh, 2020-01-28:
Hi, the supported file format is aot file, could you run wamrc like:
wamrc -o test.aot test.wasm
Or
wamrc -o test.aot --format=aot test.wasm
and test again?

masomel, 2020-01-28:
I did also test running my code compiled with the `--format=aot` flag, and it ran with no issues.

However, given that wamrc produces ELF object files, I am wondering if there is a plan to add support for running AoT-compiled ELF object files in WAMR in the future?

wenyongh, 2020-02-04:
Thanks a lot for the suggestion, currently no plan yet, loading ELF file requires the support of system dynamic loader, e.g. using dlopen() function to load a shared library (.so) file, which normally isn't provided by embedded OS. It may be a good idea for Linux, we will discuss it internally.

masomel, 2020-02-04:
Sounds good, thank you!

wenyongh, 2020-03-27:
AOT is supported in linux-sgx now, so close the issue.

closed 2020-03-27 by wenyongh