wasm-micro-runtime/wasm-micro-runtime #163: unrecognized compiler option ‘-mindirect-branch-register’

jxiao1, 2020-02-12:
In my local system, when I try to build the wamr or wamr-sdk and so on, there is the following error:
[  2%] Building C object CMakeFiles/vmlib.dir/home/junxian.xjx/workspace/wasm-micro-runtime/core/shared/platform/linux/bh_assert.c.o
cc: error: unrecognized command line option ‘-mindirect-branch-register’
make[2]: *** [CMakeFiles/vmlib.dir/home/junxian.xjx/workspace/wasm-micro-runtime/core/shared/platform/linux/bh_assert.c.o] Error 1
make[1]: *** [CMakeFiles/vmlib.dir/all] Error 2
make: *** [all] Error 2

I just workaround it like this:
diff --git a/wamr-compiler/CMakeLists.txt b/wamr-compiler/CMakeLists.txt
index ed2198c..c349f88 100644
--- a/wamr-compiler/CMakeLists.txt
+++ b/wamr-compiler/CMakeLists.txt
@@ -97,9 +97,9 @@ include (${IWASM_DIR}/compilation/iwasm_compl.cmake)

 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Wextra -Wformat -Wformat-security")
 # set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wconversion -Wsign-conversion")
-if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
-set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
-endif ()
+#if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
+#set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
+#endif ()
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fstack-protector-strong --param ssp-buffer-size=4")
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wl,-z,noexecstack,-z,relro,-z,now")

xujuntwt95329, 2020-02-12:
Hi,

Are you using Mac OS, seems your default compiler is `clang`, and *-mindirect-branch-register* only works on GCC.

The `product-mini/platforms/darwin` is prepared for Mac OS and you can build it without any modification.
As for the wamr-compiler, yes, currently the CMakeLists file is only for linux/GCC, to build it with clang, we need to comment line 101:
``` cmake
# set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
```
and set `WAMR_BUILD_PLATFORM` to "darwin" at line 7:
``` cmake
set (WAMR_BUILD_PLATFORM "darwin")
```

wenyongh, 2020-02-18:
We have enhanced the CMakeLists.txt to check the C compiler, if it is clang, the flag -mindirect-branch-register will be disabled by default.

closed 2020-02-25 by wenyongh