wasm-micro-runtime/wasm-micro-runtime #165: cannot copy libraries when use linux-sgx as subdirectory in other projects

jxiao1, 2020-02-13:
If I use this repo as a submodule in another project and build linux-sgx like this:
add_subdirectory(wasm/wamr/product-mini/platforms/linux-sgx/)

The following copy action will result in error:
set (copy_libs_cmd cp -a libvmlib.a libextlib.a ../enclave-sample/)
add_custom_target (copy_libs_to_enclave ALL COMMAND ${copy_libs_cmd} DEPENDS vmlib extlib)

I just removed the above "set" line as a workaround

jxiao1, 2020-02-13:
Another problem is I need to source <SGX_SDK dir>/environment before build linux-sgx,  I suggest to add a default SDX_SDK path value in the linux-sgx/CMakelist.txt file.

xwang98, 2020-02-17:
@jxiao1 thank you for fixing the problem.

closed 2020-02-17 by xwang98