wasm-micro-runtime/wasm-micro-runtime #170: Error when try to cross compile "simple" example for ARM

oubotong, 2020-02-22:
In the directory `samples/simple`, I modified the configuration `wamr_config_simple.cmake` and set the target to `ARMV5` which is supported for the WASM VM core. Then I manually changed the compiler for `build.sh` in the same directory and `build_sdk.sh` in `wamr_sdk` directory to `aarch*` which is the right compiler for cross compiling. Then I met with these error:
```#####################build wamr sdk
CMAKE_DEXTRA_SDK_INCLUDE_PATH=
User config file: [/home/richard/wasm-micro-runtime/samples/simple/wamr_config_simple.cmake]
#########################################################
profile /home/richard/wasm-micro-runtime/wamr-sdk/out/simple already exists
  skip the build process and use the previous settings: [y]
  or delete the profile and generate a new one:          n
N

##############  Start to build wasm app sdk  ###############
using builtin libc toolchain
-- The C compiler identification is Clang 9.0.0
-- The CXX compiler identification is Clang 9.0.0
-- Check for working C compiler: /opt/wasi-sdk/bin/clang
-- Check for working C compiler: /opt/wasi-sdk/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/wasi-sdk/bin/clang++
-- Check for working CXX compiler: /opt/wasi-sdk/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
WAMR_BUILD_APP_FRAMEWORK
Add app module base
Add app module connection
Add app module sensor
-- Configuring done
-- Generating done
-- Build files have been written to: /home/richard/wasm-micro-runtime/wamr-sdk/app/build
Scanning dependencies of target app_framework
[ 12%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/app-native-shared/attr_container.c.o
[ 25%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/app-native-shared/restful_utils.c.o
[ 37%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/base/app/bh_platform.c.o
[ 50%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/base/app/request.c.o
[ 62%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/base/app/timer.c.o
[ 75%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/connection/app/connection.c.o
[ 87%] Building C object CMakeFiles/app_framework.dir/home/richard/wasm-micro-runtime/core/app-framework/sensor/app/sensor.c.o
[100%] Linking C static library libapp_framework.a
[100%] Built target app_framework
Successfully built app-sdk under /home/richard/wasm-micro-runtime/wamr-sdk/out/simple/app-sdk

##############  Start to build runtime sdk  ###############
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/richard/optee/toolchains/aarch64/bin/aarch64-linux-gnu-g++
-- Check for working CXX compiler: /home/richard/optee/toolchains/aarch64/bin/aarch64-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Add native module base
Add native module connection
Add native module sensor
-- Build Configurations:
     Build as target ARMV5
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR JIT disbled
     Libc builtin enabled
     Libc WASI disbled
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/richard/wasm-micro-runtime/wamr-sdk/runtime/build_runtime_sdk
Scanning dependencies of target vmlib
[  1%] Building C object CMakeFiles/vmlib.dir/home/richard/wasm-micro-runtime/core/shared/platform/linux/bh_assert.c.o
cc: error: unrecognized command line option ‘-marm’; did you mean ‘-mabm’?
CMakeFiles/vmlib.dir/build.make:62: recipe for target 'CMakeFiles/vmlib.dir/home/richard/wasm-micro-runtime/core/shared/platform/linux/bh_assert.c.o' failed
make[2]: *** [CMakeFiles/vmlib.dir/home/richard/wasm-micro-runtime/core/shared/platform/linux/bh_assert.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/vmlib.dir/all' failed
make[1]: *** [CMakeFiles/vmlib.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Failed to build runtime sdk

``` 
My question is that is there a way to cross compile the samples for ARM. In the tutorial, it only mentions how to cross compile the wasm VM core. 
Thanks for help.

xwang98, 2020-02-22:
@oubotong a good solution for cross compilation is using the toolchain file argument of cmake. we will add a sample for cross compilation of the app framework.

oubotong, 2020-02-22:
Thanks, I thought I missed some relevant documents for cross compiling the samples. I will figure out how to make the demo work.

xujuntwt95329, 2020-02-22:
@oubotong 
According to the output of the cmake, seems you only pass the CMAKE_CXX_COMPILER to the CMakeLists file, and the C compiler is still GNU cc, which don't support the `-marm` option
```
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
```
WAMR is written in C, you may need `aarch64-linux-gnu-gcc`

To fix this quickly, you can use -DCMAKE_C_COMPILER=/path/to/your/cross-compiler to pass the C compiler to cmake in both `build.sh` under `sample/simple` and `build_sdk.sh` in `wamr-sdk`

In addition, some ARM cross compiler doesn't support the `-m32` option, if you encounter this problem, just comment the line 60 in `build-scripts/config_common.cmake` and try again:
``` cmake
# add_definitions (-m32)
```

Hope this can help you.

oubotong, 2020-02-22:
My mistake, I forgot to modify a few lines of commands. And I also need to use `arm-linux-gnueabihf*` instead of the ones for AARCH64. However, another problem comes out and I'm trying to fix it.
```
./build_sdk.sh: line 212: cd: /home/oubotong/wasm-micro-runtime-arm/wamr-sdk/out/simple/app-sdk/wamr-app-framework/share: No such file or directory
#####################build simple project
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oubotong/wasm-micro-runtime-arm/samples/simple/cmake_build
Scanning dependencies of target simple
[ 25%] Building C object CMakeFiles/simple.dir/src/iwasm_main.c.o
In file included from /home/oubotong/wasm-micro-runtime-arm/samples/simple/src/iwasm_main.c:26:0:
/home/oubotong/wasm-micro-runtime-arm/samples/simple/../../wamr-sdk/out/simple/runtime-sdk/include/app_manager_export.h:10:10: fatal error: bi-inc/shared_utils.h: No such file or directory
 #include "bi-inc/shared_utils.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/simple.dir/build.make:86: recipe for target 'CMakeFiles/simple.dir/src/iwasm_main.c.o' failed
make[2]: *** [CMakeFiles/simple.dir/src/iwasm_main.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/simple.dir/all' failed
make[1]: *** [CMakeFiles/simple.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
BUILD_FAIL simple exit as 0\n

```

oubotong, 2020-02-22:
I solve this problem by useing `CC=arm-linux-gnueabihf-gcc CXX=arm-linux-gnueabihf-g++` instead of  using `-DCMAKE_C_COMPILER`.

xwang98, 2020-02-22:
@oubotong the recently patch has make the simple sample easier to build for multiple profiles:

https://github.com/bytecodealliance/wasm-micro-runtime/tree/master/samples/simple#build-the-sample

oubotong, 2020-02-22:
Great!! Thanks! I also figured it out yesterday.

closed 2020-02-23 by xwang98