wasm-micro-runtime/wasm-micro-runtime #175: Build error on macos

renjipanicker, 2020-02-23:
I cloned the head of master branch and tried to build 'products/platforms/darwin' on macos.
I get the following warning and error:

```
Renjis-MacBook-Pro:build_wamr renji$ make
Scanning dependencies of target libiwasm
[  1%] Building C object CMakeFiles/libiwasm.dir/Users/renji/tmp/wasm-micro-runtime/core/shared/platform/darwin/bh_assert.c.o
[  3%] Building C object CMakeFiles/libiwasm.dir/Users/renji/tmp/wasm-micro-runtime/core/shared/platform/darwin/bh_definition.c.o
/Users/renji/tmp/wasm-micro-runtime/core/shared/platform/darwin/bh_definition.c:8:9: warning: 'RSIZE_MAX' macro redefined
      [-Wmacro-redefined]
#define RSIZE_MAX 0x7FFFFFFF
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/stdint.h:176:9: note: 
      previous definition is here
#define RSIZE_MAX         (SIZE_MAX >> 1)
        ^
1 warning generated.
[  5%] Building C object CMakeFiles/libiwasm.dir/Users/renji/tmp/wasm-micro-runtime/core/shared/platform/darwin/bh_platform.c.o
/Users/renji/tmp/wasm-micro-runtime/core/shared/platform/darwin/bh_platform.c:142:34: error: use of undeclared identifier
      'MADV_HUGEPAGE'
        if (!madvise(addr, size, MADV_HUGEPAGE))
                                 ^
1 error generated.
make[2]: *** [CMakeFiles/libiwasm.dir/Users/renji/tmp/wasm-micro-runtime/core/shared/platform/darwin/bh_platform.c.o] Error 1
make[1]: *** [CMakeFiles/libiwasm.dir/all] Error 2
make: *** [all] Error 2
```

CMAKE files were generated as follows:
```
Renjis-MacBook-Pro:build_wamr renji$ cmake ../wasm-micro-runtime/product-mini/platforms/darwin/
-- The C compiler identification is AppleClang 11.0.0.11000033
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build Configurations:
     Build as target X86_64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT enabled
     WAMR JIT disbled
     Libc builtin enabled
     Libc WASI disbled
-- The ASM compiler identification is Clang
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/renji/tmp/build_wamr
```
CLANG version is:
```
Renjis-MacBook-Pro:build_wamr renji$ clang --version
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin19.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

OS is:
`macOS Catalina, version 10.15.3`

xwang98, 2020-02-24:
@renjipanicker @jondong by any chance, could you help to identify the root cause and fix it? We don't have MacOS development environment at this moment.

renjipanicker, 2020-02-24:
Done, pull request submitted.

xwang98, 2020-02-24:
Awesome! @renjipanicker

closed 2020-02-24 by xwang98