wasm-micro-runtime/wasm-micro-runtime #177: Functions wasm_application_execute_main and wasm_application_execute_func should take wasm_exec_env_t as first parameter

renjipanicker, 2020-02-23:
This is a feature request:
Functions `wasm_application_execute_main`, `wasm_application_execute_func`, `wasm_runtime_set_custom_data`, and `wasm_runtime_get_custom_data` should take `wasm_exec_env_t` as first parameter, instead of `wasm_module_inst_t`.

The custom data should be stored in the `wasm_exec_env_t` object, instead of in `wasm_module_inst_t`.

This will allow us to preload the modules and keep them in memory for repeated use.

And more important, it will allow us to set a different custom-data on a per-call basis.

This will become necessary when multithreading is implemented in future.

wenyongh, 2020-02-25:
Hi @renjipanicker, thanks for the good suggestions!

For the API wasm_application_execute_main() and wasm_application_execute_func(), it will create exec_env internally and then call wasm_runtime_call_wasm() with exec_env argument. If developer runs the API in different threads, different exec_env will be created, so it should be unnecessary to replace the argument with module instance.

For the custom data, maybe we can keep the API wasm_runtime_set_custom_data(module_inst, custom_data), and add new APIs for exec_env, e.g.:

    void wasm_exec_env_set_custom_data(wasm_exec_env_t exec_env, void *custom_data);
    void *wasm_exec_env_set_custom_data(wasm_exec_env_t exec_env);

renjipanicker, 2020-02-25:
@wenyongh happy to help - thanks for a great product!

The use case I have in mind is something like this (untested code):
```
void exec(const char* mod_name, user_data* ud, int argc, const char* argv[]){
    wasm_module_inst_t inst = get_or_add_module_to_cache(mod_name); // if previously loaded, return cached module, else load and add to cache
    wasm_exec_env_t env = wasm_runtime_create_exec_env(inst, 8*1024);
    wasm_exec_env_set_custom_data(env, ud); // proposed function, now ud can be used in native functions
    wasm_application_execute_main(env, argc, fargv)
}

```

wenyongh, 2020-05-15:
Two APIs were added:
 
    wasm_runtime_set_user_data()
    wasm_runtime_get_user_data()

@renjipanicker the argument passed to wasm_application_execute_main() is module_inst, but not exec_env, the function wasm_application_execute_main() creates exec_env internally and uses it to call function wasm_runtime_call_wasm() to the call main function. So you cannot set user data to exec_env when calling wasm_application_main(). If you want to execute main, and set user data to exec_env, you might:

      call wasm_runtime_lookup_function(module_inst, "main", NULL) to get the main function instance
      create exec env
      call wasm_runtime_set_user_data(exec_env, user_data)
      prepare arguments for main if needed
      call wasm_runtime_call_wasm(exec_env, func, argc, argv) to call main function

And then in the native wrapper function, you can get user data:
    user_data = wasm_runtime_set_user_data(exec_env)

closed 2020-06-12 by wenyongh