wasm-micro-runtime/wasm-micro-runtime #184: Llvm clone depth

Rochet2, 2020-03-03:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/2e36149e326dab2374eda01bd7be5810942a5db5/wamr-compiler/build_llvm.sh#L11

Would it be a good idea to restrict clone depth to 1 or similar in the build script?
Downloading all history (200 000 commits) takes forever and is probably unnecessary most of the time I would imagine.

xwang98, 2020-03-05:
That would be a good idea. Do you know any command/arugment available for downloading the snapshot without history versions? @Rochet2

Rochet2, 2020-03-05:
One can use `git clone --depth 1 https://github.com/llvm-mirror/llvm.git`
to only get latest commit. Or specify another depth to get more if needed.
Afterwards one can use `git fetch --unshallow` to fetch all history.

xwang98, 2020-03-05:
Thank you for the comments. I just updated the script. @Rochet2

closed 2020-03-05 by xwang98