wasm-micro-runtime/wasm-micro-runtime #190: Multiple modules and sharing?

Rochet2, 2020-03-07:
In the current implementation..
1. Is loading multiple modules supported at this moment?
2. Can I share tables between them?
3. Can I share memory between them?

If not, then is it possible to avoid sharing of the table somehow? It seems to be shared by default in my test, unless I made some error.

I was able to load multiple modules by calling `wasm_runtime_load` and `wasm_runtime_instantiate` twice. Afterwards I tried calling the `run` function.
It seems that the runtime created a table on its own without me creating it in the modules or in C explicitly.
The table was also shared, so the call_indirect succeeded. However when I tested to share memory it did not succeed in the same way.

```C
    char error_buf1[128] = { '\0' };
    wasm_module_t module1 = wasm_runtime_load(wasm1, size1, error_buf1, sizeof(error_buf1));
    wasm_module_inst_t module_inst1 = wasm_runtime_instantiate(module1, stack_size, heap_size, error_buf1, sizeof(error_buf1));
    char error_buf2[128] = { '\0' };
    wasm_module_t module2 = wasm_runtime_load(wasm2, size2, error_buf2, sizeof(error_buf2));
    wasm_module_inst_t module_inst2 = wasm_runtime_instantiate(module2, stack_size, heap_size, error_buf2, sizeof(error_buf2));
    
    wasm_function_inst_t func = wasm_runtime_lookup_function(module_inst1, "run", "()i32");
    wasm_exec_env_t exec_env = wasm_runtime_create_exec_env(module_inst2, stack_size);
    uint32 argv[] = {0};
    if (wasm_runtime_call_wasm(exec_env, func, 0, argv))
        printf("function return: %d\n", argv[0]);

```
```wat
(module
  (import "C" "table" (table 1 funcref))
  (type $multiply2Type (func (param i32) (result i32)))
  (func (export "run") (result i32)
   i32.const 33 ;; pass value to called function
   i32.const 0 ;; table index of function to call
   call_indirect (type $multiply2Type))
)
```
```wat
(module
  (import "C" "table" (table 1 funcref))
  (elem (i32.const 0) $multiply2)
  (func $multiply2 (param i32) (result i32)
   local.get 0
   local.get 0
   i32.add)  ;; multiply by 2, leave result on stack
)
```

Rochet2, 2020-03-10:
I believe all my multiple module loading problems were solved with the latest allocator changes.
By using the following code from the linux example the modules no longer share tables.
```cpp
        init_args.mem_alloc_type = Alloc_With_Allocator;
        init_args.mem_alloc.allocator.malloc_func = malloc;
        init_args.mem_alloc.allocator.realloc_func = realloc;
        init_args.mem_alloc.allocator.free_func = free;
```

**As such the only question at this time that stands is if one can actually create multiple runtimes / modules and run them separately of each other safely.**

By the way. Great job on the API. It is exactly what I need, is clear and simple to use.

wenyongh, 2020-03-12:
@Rochet2, thanks for using the new API and giving feedback! Have you used the latest version? We have changed the API to
     wasm_runtime_full_init()
     wasm_runtime_init()
Developer can use one of them to init. And use
    wasm_runtime_destroy()
to destroy, no need to call hb_memory_destroy() or wasm_runtime_memory_destroy() again. This really makes it more simple to use.
And now the field name should be: init_args.mem_alloc_option, but not init_args.mem_alloc.

For the multiple module instances support, now we haven't supported shared memory/table yet, each module instance has its single memory/table. If you create two module instances with the same module, and run them in different threads, it should work. If you create one module instance, and create two exec_env to run same module in different threads, it may fail, e.g. memory.grow will reallocate the memory space and cause the other thread accessing invalid original memory space. And if one thread destroys module instance, the other thread will access invalid data.

Rochet2, 2020-03-12:
I am now using latest version.
Thank you for confirming that multiple instances should work.
When trying to use Alloc_With_Pool I did get errors related to memory access when running multiple modules (single thread), however using Alloc_With_Allocator works just fine.
I assume Alloc_With_Pool is not intended to be used with multiple modules.

By multiple modules I mean creating separate module, instance and execution environment for each  and then calling functions in them.

closed 2020-03-12 by Rochet2