wasm-micro-runtime/wasm-micro-runtime #200: Multi module support

enleeten, 2020-03-15:
Is it possible to load multiple modules and import/export functions between them?

If so, generally how does that flow work?

Thanks!

xwang98, 2020-03-15:
@enleeten It is also potential requirements by our products that use WAMR. We will evaluate it and have further discussion with you.

enleeten, 2020-03-15:
Awesome, thanks.

Really cool project! I'm working on some tech that compiles to wasm and have mainly been using the reference interpreter and have played around with graalwasm a bit.

Rochet2, 2020-04-27:
As a workaround we created multiple instances of the runtime and communicate between them through exposing host functions that allow calling functions from another module. Essentially it achieves the same. We can call functions and pass data between modules, but probably somewhat inefficiently. With some modifications to WAMR sharing of memory might also be possible to implement to speed up passing large strings from one module to another.

xwang98, 2020-04-27:
@Rochet2 Yes, basically it is how we are going to implement. Currently there is no buffer sharing between modules since two modules are two sandboxes. Yes, we should look for a way to share the buffer among modules without breaking the basic isolation.

scottlin2019, 2020-06-01:
> @Rochet2 Yes, basically it is how we are going to implement. Currently there is no buffer sharing between modules since two modules are two sandboxes. Yes, we should look for a way to share the buffer among modules without breaking the basic isolation.

@xwang98 when will support multi modules? Do you have a plan? thanks!

wenyongh, 2020-06-12:
Multi module is supported in interpreter mode now, seems that we can close the issue.

closed 2020-06-12 by wenyongh