wasm-micro-runtime/wasm-micro-runtime #212: Segmentation fault and odd behaviour

Rochet2, 2020-03-25:
I am running into problems running various programs.
There is odd behaviour and crashing when running some programs.
For example I am receiving errors about:
- Segmentation fault
- Stack overflow, even if my code contains no recursion
- Converting string to number always returns 0, for example for the string "99"

Details:
- Using the linux example to run
- The problem occurs on interp, fast interp. Did not test AOT.
- I am using the commit 01e85144f4063db9809464fb40c0a95b16f83562

Example settings for cmake and running:
```
cmake .. -DWAMR_BUILD_AOT=0 -DWAMR_BUILD_FAST_INTERP=0 -DWAMR_BUILD_LIBC_BUILTIN=0 -DWAMR_BUILD_LIBC_WASI=0
```

Here is a small test `.wat` file that results in a `Segmentation fault (core dumped)` error.
I added the `.wasm` file of this code as an attachment here: [main4.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/4379820/main4.zip).
Commenting out either the export for `__data_end` or the export for `__heap_base` will allow the program to run.
You can also change one of the underscores to `a` or add an underscore to make it run.
For example `a_heap_base` and `___heap_base` respectively.
```wat
(module
  (memory (;0;) 0)
  (global (;0;) i32 (i32.const 0))
  (global (;1;) i32 (i32.const 1))
  (func $main (result i32)
    i32.const 20)
  (export "memory" (memory 0))
  (export "main" (func $main))
  (export "__data_end" (global 0))
  (export "__heap_base" (global 1))
)
```

GDB report:
```
Starting program: /home/local/<user>/Documents/wasm-micro-runtime/product-mini/platforms/linux/build/iwasm main4.wasm
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000555555569a86 in load_from_sections (module=0x5555557782f0, sections=0x5555557783c0, 
    error_buf=0x7fffffffd930 "", error_buf_size=128)
    at /home/local/<user>/Documents/wasm-micro-runtime/core/iwasm/interpreter/wasm_loader.c:1630
1630                            if (global->type == VALUE_TYPE_I32
(gdb) 
```

If you need more information or need me to try something do tell.

wenyongh, 2020-03-27:
@Rochet2, thanks for finding the issue! The exported globals __data_end and __heap_base have special meanings, generally they are generated by clang compiler to mark the end of data, the end of explicit stack, and also the beginning of heap area which is used by memory allocation. We resolve these data to try to decrease memory usage, and prepare for future multi-thread support. If the __heap_base is detected and there is no memory.grow opcode, which means no memory allocation and heap area isn't used, we will shrink the linear memory space to the value of __heap_base to decrease memory usage, which is very important for embedded system.

There is issue when resolving these data, we have fixed it in latest patch, please try again.

Rochet2, 2020-03-27:
Thanks for quick response and fix.
The segmentation fault was fixed and at the same time the functionality in my code for converting a string to number started working.

I still run into the stack overflow, but I will make a separate issue for that once I get information on how to reproduce it.

closed 2020-03-27 by Rochet2