wasm-micro-runtime/wasm-micro-runtime #213: Standalone executable that loads wasm module compiled from Assemblyscript sample

rahul-thakoor, 2020-03-26:
Hello,

Is it possible to use the Embedding API to create a standalone binary for linux that loads a wasm module compiled from assemblyscript using your provided [wamr-app-lib](https://github.com/bytecodealliance/wasm-micro-runtime/tree/master/assembly-script/wamr_app_lib) ? My CMakeLists.txt file is similar to [this](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/samples/basic/CMakeLists.txt).
Would setting `WAMR_BUILD_APP_FRAMEWORK=1` suffice?

Is it possible to use https://github.com/jedisct1/as-wasi? or do I need to stick to the wamr-app-lib?

Thanks

xwang98, 2020-03-26:
@rahul-thakoor To support the app-lib, please also add:
`set (WAMR_BUILD_APP_LIST WAMR_APP_BUILD_BASE)`

> Is it possible to use https://github.com/jedisct1/as-wasi? or do I need to stick to the wamr-app-lib?
Yes, we can use it. No need to stick to wamr-app-lib

rahul-thakoor, 2020-03-27:
Thanks for the quick response. I used 
```
set (WAMR_BUILD_APP_FRAMEWORK 1)
set (WAMR_BUILD_APP_LIST WAMR_APP_BUILD_BASE)
```
but I keep getting 

```
[02:05:59:433 - 2ACF0B80]: failed to check signature '($*)i' and resolve pointer params for import function (env printf)

[02:05:59:439 - 2ACF0B80]: warning: fail to link import function (env, printf)

```
Should I be registering the native symbols manually? is there a quick way to do this?
I am trying to customise the basic sample app to load the wasm module compiled from assemblyscript. 

Thanks

xujuntwt95329, 2020-03-27:
Our new [register native symbol API](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/export_native_api.md) has enabled automatic signature checking, and printf is defined as ($*)i, which means two args are required. Seems the `printf` imported by assemblyscript doesn't match this signature.

If you are using the console lib from `assembly-script/wamr_app_lib`, PR #215 will solve the problem.

Thanks

rahul-thakoor, 2020-03-27:
thanks that resolved the issue for the `printf` import

However, when using latest version of assemblyscript and as-wasi which now uses wasi_snapshot_preview1, it fails with:
`warning: fail to link import function (wasi_snapshot_preview1, fd_write)`

i'm able to use it by using older version where `as-wasi`

xujuntwt95329, 2020-03-27:
The [wasi-sdk](https://github.com/WebAssembly/wasi-sdk/releases) currently import wasi APIs from wasi_unstable, so we use `wasi_unstable` as the module name.

As a workaround, you can change the `wasi_unstable` into `wasi_snapshot_preview1` in [wasm_native.c](https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/core/iwasm/common/wasm_native.c#L227)
``` c
#if WASM_ENABLE_LIBC_WASI != 0
    n_native_symbols = get_libc_wasi_export_apis(&native_symbols);
    if (!wasm_native_register_natives("wasi_snapshot_preview1",
                                      native_symbols, n_native_symbols))
        return false;
#endif
```

I haven't tested it on `as-wasi` library, but it should work as long as the symbol name and signature of the wasi API imported by `as-wasi` is not changed.

Hope this can help you.

Thanks

rahul-thakoor, 2020-03-30:
Thanks a lot for the prompt and detailed response. I will try it. Cheers

wenyongh, 2020-04-07:
@rahul-thakoor the wasi symbols array with module name "wasi_snapshot_preview1" and "wasi_unstable" are both registered to runtime by default, so you don't need to register it again in wasm_native.c now.

rahul-thakoor, 2020-04-07:
awesome! thanks a lot.

wenyongh, 2020-07-07:
Seems that we can close the issue.

closed 2020-07-07 by wenyongh