wasm-micro-runtime/wasm-micro-runtime #216: Unable to use Rust regex library

Rochet2, 2020-03-27:
I am using the commit 75009a242128ee4fb49c3cb7e73c43198ef17daf

I am trying to create a wasm module with rust.
The module uses a rust regex library that seems to work fine on other runtimes such as wasmtime and wasmer, so I assume the issue is not with the module but the runtime.
When creating a new Regex object I immediately get the following error:
```
$ ./iwasm release.wasm
foo
thread '<unnamed>' panicked at 'internal error: entered unreachable code: not all instructions were compiled! found uncompiled instruction: Compiled(Bytes(InstBytes { goto: 1, start: 98, end: 98 }))', /home/<user>/.cargo/registry/src/github.com-1ecc6299db9ec823/regex-1.3.6/src/compile.rs:766:18
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
Runtime exception: Exception: unreachable
```

The code is the following:
```rust
use regex::Regex;

#[no_mangle]
pub extern "C" fn main() {
    println!("foo");
    let result = Regex::new("abc");
    println!("bar");
}
```

The expected output would be
```
$ ./iwasm debug.wasm
foo
bar
```

Here is the program code in total and a ready built wasm that fails to run as well as a working one: 
[repro.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/4393555/repro.zip)
Because the issue is dependent on how the module is built I included the errorneous `release.wasm` file to the zip and a working `debug.wasm`

You can build the repro program by installing [rust](https://www.rust-lang.org/learn/get-started), running `cargo install wasm-pack`, installing [cargo-wasi](https://github.com/bytecodealliance/cargo-wasi#installation) and then running `cargo wasi build --release` inside the folder with the `Cargo.toml` file. To build the debug version you would run `cargo wasi build`.
The generated wasm files should be in `target/wasm32-wasi/release` and `target/wasm32-wasi/debug` respectively.
Note that a debug build of the module seems to work fine and only release build has issues.

**Micro runtime does not support `wasi_snapshot_preview1`** and I could not find a way to build the wasm with `wasi_unstable` wich is available. Some options I thought of were to just change the strings in the wasm from wasi_snapshot_preview1 to wasi_unstable and editing the runtime. For now I decided to edit the runtime and you can find `WAMR_wasi.diff` inside the zip that contains my changes. You can apply it to the runtime code after cloning WAMR with `git apply WAMR_wasi.diff`.

Unfortunately for now I have only had the issue with regex library so far and converting the wasm file to wat results in million lines of code making it hard to pinpoint the issue.
Suggestions on what to try are welcome.

wenyongh, 2020-04-07:
@Rochet2, thanks a lot for finding the issue! Now we have enabled the support for wasi_snapshot_preview1, patch has been merged, could you please pull and try again?

Rochet2, 2020-04-09:
Seems both succeed now.
Using linux mini project with default cmake settings on commit 5e196253f66e4d6605cb1ed2c1c0636ccb51e448
```bash
$ ./iwasm debug.wasm 
foo
Exception: WASM interp failed: stack overflow.
$ ./iwasm --stack-size=1000000 debug.wasm 
foo
bar
$ ./iwasm release.wasm 
foo
bar
```

closed 2020-04-09 by Rochet2