wasm-micro-runtime/wasm-micro-runtime #219: format strings for fixed size hould use fixed size format string macros

kfessel, 2020-03-31:
especially instead of buffer created format-strings 

    --- a/core/iwasm/common/wasm_runtime_common.c
    +++ b/core/iwasm/common/wasm_runtime_common.c
    @@ -1374,15 +1374,10 @@ wasm_application_execute_func(WASMModuleInstanceCommon *module_inst,
                break;
            case VALUE_TYPE_I64:
            {
    -            char buf[16];
                union { uint64 val; uint32 parts[2]; } u;
                u.parts[0] = argv1[0];
                u.parts[1] = argv1[1];
    -            if (sizeof(long) == 4)
    -                snprintf(buf, sizeof(buf), "%s", "0x%llx:i64");
    -            else
    -                snprintf(buf, sizeof(buf), "%s", "0x%lx:i64");
    -            os_printf(buf, u.val);
    +            os_printf("0x%"PRIx64":i64", u.val);
                break;
            }

wenyongh, 2020-04-03:
@kfessel thanks for the suggestion! Not sure whether all the platforms have this macro, seems it is better to check macro firstly to avoid compile error in some platform:

    #ifdef PRIx64
        os_printf("0x%"PRIx64":i64", u.val);
    #else
        use the original code
    #end

kfessel, 2020-04-03:
Since this is part of the C99 standard library and it is easy to implement i think most platforms support it.
lets do a quick check :
alios: gcc libc (+) or arm libc (+)
zephyr: minimal-libc(+) or newlibc (+)
since darwin is a full unix os using llvm i assume it has it in its libc
linux glibc musl  have it
androids boinc has it (copy from OpenBSD)
vwworks: windriver says they support C11 (i think this include C99)

even the avrlibc has it (since >12years)
microsoft was a bit late adding it 2015 since they focused on C++ 
according to https://www.cplusplus.com/reference/cinttypes/ these macros are also part of C++11

and if one needs to provide it there is the BSD licensed version from OpenBSD
https://android.googlesource.com/platform/bionic/+/refs/tags/android-wear-9.0.0_r14/libc/include/inttypes.h

but at this specific place the code currently in master is testing the compilers ability to deep dive into the code. The original should be:  ( at least give the compiler a chance to throw away the unneeded code)

        case VALUE_TYPE_I64:
        {
            char * buf;
            union { uint64 val; uint32 parts[2]; } u;
            u.parts[0] = argv1[0];
            u.parts[1] = argv1[1];
            if (sizeof(long) == 4){
                buf = "0x%llx:i64";
            }else{
                buf = "0x%lx:i64";
            }
            os_printf(buf, u.val);
            break;
        }

wenyongh, 2020-04-07:
Yes, thanks a lot for listing the platforms, seems that most platforms support the macro PRIx64, but there are really some special platforms used by some users, which don't define the macro. Some of them don't have most of the header files  included in platform/linux/platform_internal.h, including inttype.h, and even don't have float/double type and many libc functions, many of the APIs are customized. So we prefer to add #ifdef to control it, the code is modified as:

    #ifdef PRIx64
            os_printf("0x%"PRIx64":i64", u.val);
    #else
            char buf[16];
            if (sizeof(long) == 4)
                snprintf(buf, sizeof(buf), "%s", "0x%llx:i64");
            else
                snprintf(buf, sizeof(buf), "%s", "0x%lx:i64");
            os_printf(buf, u.val);
    #endif

kfessel, 2020-04-08:
Even though i prefer the standard libc way (and would like to see the OpenBSD header included for incomplete platforms as you do with the free BSD Math functions). I would like also like to see your version improved:
It would be better for the non macro version to instead of using snprintf assigning a buffer pointer to a constant value in by that not create a extra buffer and copy data from flash (or programm-memory) to ram (stack).

    #ifdef PRIx64
            os_printf("0x%"PRIx64":i64", u.val);
    #else
        char * buf;
        if (sizeof(long) == 4){
            buf = "0x%llx:i64";
        }else{
            buf = "0x%lx:i64";
        }
        os_printf(buf, u.val);
    #endif

P.S.: I would also like see the inttype.h types  used throughout the Project since my editor has build in highlighting for them and not for your types.
stdint types like: uint8_t 
(there are even special versions (least and fast which provide integer sizes optimized for the need at that special position (hey compiler i need at least 8 bit but i like it to be fast so chose a bigger int if you think its faster))

https://en.cppreference.com/w/cpp/types/integer

wenyongh, 2020-05-15:
Some user's platform really doesn't have macro PRIx64, and they don't want to implement that. We prefer to keep the source code unchanged. And if user wants, he can also define the macro in his platform and enhance the related printf functions to support the format.

closed 2020-05-15 by wenyongh