wasm-micro-runtime/wasm-micro-runtime #226: Changing product-mini/android WAMR_BUILD_TARGET does not work

Rochet2, 2020-04-02:
## Problem
Using commit ef5ceffe71ee3a9429e72b76f1e4d94c26d90249 with `WAMR_BUILD_TARGET` set to `X86_64` fails with message `Build target isn't set` even if based on the CMake code it should work.

## Suggested fix
Using `"${WAMR_BUILD_TARGET}" STREQUAL "X86_64"` instead of `${WAMR_BUILD_TARGET} STREQUAL "X86_64"` seems to fix the issue.

## Details
I changed the build target and added some debug code to see what `WAMR_BUILD_TARGET` is and if the expected equality check succeeds or not.

The output shows that the variable is set correctly (two identical `aX86_64a` prints), but the test fails 
 and prints `NOTWORKS` due to unknown reason.

The suggested fix of adding quotes will make `STREQUAL` work fine and `WORKS` is printed.

```diff
diff --git a/core/iwasm/common/iwasm_common.cmake b/core/iwasm/common/iwasm_common.cmake
index 3e2acfd..ba1cf33 100644
--- a/core/iwasm/common/iwasm_common.cmake
+++ b/core/iwasm/common/iwasm_common.cmake
@@ -10,6 +10,15 @@ add_definitions(-DBH_FREE=wasm_runtime_free)
 
 file (GLOB c_source_all ${IWASM_COMMON_DIR}/*.c)
 
+MESSAGE("a${WAMR_BUILD_TARGET}a")
+MESSAGE("aX86_64a")
+if (${WAMR_BUILD_TARGET} STREQUAL "X86_64")
+MESSAGE("WORKS")
+else()
+MESSAGE("NOTWORKS")
+endif()
+
+MESSAGE("WATT" ${WAMR_BUILD_TARGET} ${WAMR_BUILD_TARGET} STREQUAL "X86_64")
 if (${WAMR_BUILD_TARGET} STREQUAL "X86_64" OR ${WAMR_BUILD_TARGET} STREQUAL "AMD_64")
   set (source_all ${c_source_all} ${IWASM_COMMON_DIR}/arch/invokeNative_em64.s)
 elseif (${WAMR_BUILD_TARGET} STREQUAL "X86_32")
diff --git a/product-mini/platforms/android/CMakeLists.txt b/product-mini/platforms/android/CMakeLists.txt
index bdfd10e..2ead4f4 100644
--- a/product-mini/platforms/android/CMakeLists.txt
+++ b/product-mini/platforms/android/CMakeLists.txt
@@ -9,13 +9,13 @@ set (CMAKE_BUILD_TYPE       Release)
 set (CMAKE_TOOLCHAIN_FILE "$ENV{ANDROID_NDK_HOME}/build/cmake/android.toolchain.cmake")
 set (ANDROID_NDK $ENV{ANDROID_NDK_HOME})
 set (ANDROID_SDK $ENV{ANDROID_SDK_HOME})
-set (ANDROID_ABI "x86")
+set (ANDROID_ABI "x86_64")
 set (ANDROID_LD lld)
 
 project (iwasm)
 
 set (WAMR_BUILD_PLATFORM      "android")
-set (WAMR_BUILD_TARGET        "X86_32")
+set (WAMR_BUILD_TARGET        "X86_64")
 set (WAMR_BUILD_TYPE          Release)
 set (WAMR_BUILD_INTERP        1)
 set (WAMR_BUILD_AOT           0)
@@ -90,7 +90,8 @@ endif ()
 #set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fstack-protector-strong --param ssp-buffer-size=4")
 #set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wl,-z,noexecstack,-z,relro,-z,now")
 
-add_library (iwasm SHARED ${WAMR_RUNTIME_LIB_SOURCE})
+add_library (iwasm SHARED ${WAMR_RUNTIME_LIB_SOURCE} wasm-jni.cpp)
+
 if (CMAKE_BUILD_TYPE STREQUAL Release)
 target_link_libraries (iwasm ${LLVM_AVAILABLE_LIBS} -lm -ldl -landroid -llog -s)
 else()
```

Output
```
$ cmake ..
-- Build Configurations:
     Build as target X86_64
     CMAKE_BUILD_TYPE Release
     WAMR Interpreter enabled
     WAMR AOT disbled
     WAMR JIT disbled
     Libc builtin enabled
     Libc WASI disbled
     Fast interpreter disbled
aX86_64a
aX86_64a
NOTWORKS
WATTX86_64X86_64STREQUALX86_64
CMake Error at /wasm-micro-runtime/core/iwasm/common/iwasm_common.cmake:51 (message):
  Build target isn't set
Call Stack (most recent call first):
  /wasm-micro-runtime/build-scripts/runtime_lib.cmake:92 (include)
  CMakeLists.txt:72 (include)

-- Configuring incomplete, errors occurred!
See also "/wasm-micro-runtime/product-mini/platforms/android/build/CMakeFiles/CMakeOutput.log".

```

wenyongh, 2020-04-03:
@Rochet2 could you try modifying ${WAMR_BUILD_TARGET} to WAMR_BUILD_TARGET? Since there are lots of WAMR_BUILD_TARGET in scripts, but not ${WAMR_BUILD_TARGET} or "${WAMR_BUILD_TARGET}", if it is to be changed, we might need to modify many places in many files. And if WAMR_BUILD_TARGET works, we may just modify several files. Thanks a lot.

Rochet2, 2020-04-03:
Seems that using `WAMR_BUILD_TARGET` only works as well for `STREQUAL`

Code:
```
if (WAMR_BUILD_TARGET STREQUAL "X86_64")
MESSAGE("WORKS")
else()
MESSAGE("NOTWORKS")
endif()
```
Output:
```
WORKS
```

wenyongh, 2020-04-07:
@Rochet2 we have changed ${WAMR_BUILD_TARGET} to WAMR_BUILD_TARGET in latest patch, could you please test again? Thanks a lot.

Rochet2, 2020-04-09:
Seems to generate and build fine now on 5e196253f66e4d6605cb1ed2c1c0636ccb51e448.
Tested with default cmake file and also with the following changes:

```diff
diff --git a/product-mini/platforms/android/CMakeLists.txt b/product-mini/platforms/android/CMakeLists.txt
index 26f9c73..8ca07c0 100644
--- a/product-mini/platforms/android/CMakeLists.txt
+++ b/product-mini/platforms/android/CMakeLists.txt
@@ -9,13 +9,13 @@ set (CMAKE_BUILD_TYPE       Release)
 set (CMAKE_TOOLCHAIN_FILE "$ENV{ANDROID_NDK_HOME}/build/cmake/android.toolchain.cmake")
 set (ANDROID_NDK $ENV{ANDROID_NDK_HOME})
 set (ANDROID_SDK $ENV{ANDROID_SDK_HOME})
-set (ANDROID_ABI "x86")
+set (ANDROID_ABI "x86_64")
 set (ANDROID_LD lld)
 
 project (iwasm)
 
 set (WAMR_BUILD_PLATFORM      "android")
-set (WAMR_BUILD_TARGET        "X86_32")
+set (WAMR_BUILD_TARGET        "X86_64")
 set (WAMR_BUILD_TYPE          Release)
 set (WAMR_BUILD_INTERP        1)
 set (WAMR_BUILD_AOT           0)
```

```bash
$ mkdir build && cd build && cmake .. -DWAMR_RUNTIME_LIB_SOURCE=wasm-jni.cpp -DANDROID_PLATFORM=24
...
[100%] Built target iwasm
...
```

closed 2020-04-09 by Rochet2