wasm-micro-runtime/wasm-micro-runtime #227: When would I need to create a new exec_env?

Rochet2, 2020-04-03:
It seems that a function call requires exec_env.
From https://github.com/bytecodealliance/wasm-micro-runtime/issues/177 I gather that I would need new exec_env if doing some parallel processing, which is a quite clear use case.

However what about these other cases:
1. Can I use the same exec_env for multiple function calls? Or should I instead make a new exec_env for each function call?
    ```c
    wasm_runtime_call_wasm(module_inst, exec_env, function1, 0, NULL);
    wasm_runtime_call_wasm(module_inst, exec_env, function2, 0, NULL);
    ```
2. What about when doing a kind of reentrant call, where WASM calls a native function that calls a WASM function? For example here `main` calls `foo`, which at some point calls `bar`, which calls `baz` - should it create a new exec_env for the call to `baz`?
    ```c
    void bar(wasm_exec_env_t exec_env)
    {
      wasm_module_inst_t module_inst = wasm_runtime_get_module_inst(exec_env);
      wasm_function_inst_t baz = wasm_runtime_lookup_function(module_inst, "baz", NULL);
      wasm_runtime_call_wasm(module_inst, exec_env, baz, 0, NULL);
    }

    int main ()
    {
      ...
      wasm_exec_env_t exec_env = wasm_runtime_create_exec_env(module_inst, stack_size);
      wasm_function_inst_t foo = wasm_runtime_lookup_function(module_inst, "foo", NULL);
      wasm_runtime_call_wasm(module_inst, exec_env, foo, 0, NULL);
      ...
    }
    ```

xwang98, 2020-04-03:
1. yes. you can use the same `exec_env` for many calls to wasm_runtime_call_wasm().
2. please also call the wasm_runtime_call_wasm() by using the same `exec_env`  as parameter for re-entering wasm .

Your sample code has incorrect parameters for calling wasm_runtime_call_wasm(). please refer to the document below:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/embed_wamr.md#native-calls-wasm-functions-and-passes-parameters

```
  if (wasm_runtime_call_wasm(exec_env, func, 1, argv) ) {
    /* the return value is stored in argv[0] */
    printf("fib function return: %d\n", argv[0]);
  }
```

Rochet2, 2020-04-08:
Thanks for clarifying.

closed 2020-04-08 by Rochet2