wasm-micro-runtime/wasm-micro-runtime #228: Per instance import support

Hexxeh, 2020-04-06:
When using WebAssembly in a browser imports are linked per module instance: https://developer.mozilla.org/en-US/docs/WebAssembly/Using_the_JavaScript_API

It seems for wasm-micro-runtime, they're set globally and all native imports are shared between all modules. Is it a goal of this project to support the JS behaviour specified above?

xwang98, 2020-04-06:
@Hexxeh  As the standalone WASM runtime is statically compiled, it can't  support dynamically supplied import functions like JS.

Hexxeh, 2020-04-06:
Could the import functions not be part of the module instance, rather than in a global table?

I'm looking at this from the point of view of "how would you implement the WebAssembly JavaScript API using wasm-micro-runtime".

xwang98, 2020-04-07:
@Hexxeh do you mean a set of C APIs to access the elements (memory, table, global, function) exported by a WASM instance?  Currently wamr only provides the API for calling the WASM function. We should be able to access other exported elements from host runtime, but we are not clear about the use case. We are assuming the host runtime is kind of generic for running many WASM applications. In the samples from the link you pointed, each JS code is tightly bound to a pariticular WASM module. 

Could you please expain a bit more about why we probably need such API for standalone WASM runtime?

Hexxeh, 2020-04-07:
The use case would be for binding this project to a JS engine, in order to replicate the same browser APIs. In this use case, the host JS runtime is also generic and intended to run many apps, and individual apps running within that JS runtime may load and bind different WASM modules against their own JS code, proving their own imports which in turn call their JS functions.

xwang98, 2020-04-07:
It is interesting. Is it another NodeJS？ Like we can connect the wamr with JerryScript for a extremly lightwight version of NodeJS?

Hexxeh, 2020-04-07:
Yep, that's just one option, there's also Espruino and Duktape.

Hexxeh, 2020-06-17:
It seems like you can do this by registering natives before creating a module, and then just calling the destroy function. Not ideal, but it works.

Hexxeh, 2020-08-04:
It appears this is mostly addressed via the wasm-c-api additions. I think we'd want to still be able to disable the global import definitions (in order not to accidentally pollute imports, but also to avoid static globals), but everything else appears to be there! :)

closed 2020-12-10 by wenyongh