wasm-micro-runtime/wasm-micro-runtime #234: Can't respond to email

kisvegabor, 2020-04-21:
Hi,

I'm Gabor Kiss-Vamosi from LittlevGL. 

Xin Wang (@xwang98?) has contacted me yesterday but I can't respond to his email because I got "Address not found" error.  I'm writing from Gmail.

xwang98, 2020-04-21:
@kisvegabor Yes, I posted a message from lvgl website. Maybe I made a typo in my email address. I am leaving another email address wangxin-beijing@live.cn here so we may have offline discussion for the collabration opportunities of these two projects. 

We had done some experiments to have the lvgl application in wasm while the lvgl library is built into the runtime. It can greatly reduce the wasm app binary size comparing to building the whole lvgl into the wasm app. However it will require to slightly modify the lvgl header files and disable the wasm app to access the widget data structure. We have enabled a few widgets here:
https://github.com/bytecodealliance/wasm-micro-runtime/tree/master/core/app-framework/wgl

We wonder if we can have the lvgl upstream code to do small changes in order to make this process simpler. some chages like:
```
#ifndef lv_obj_t
/// the orginal lv_obj_t data structure definition
#endif 
```
then we can add such defininition in the lv_conf.h or from the compiler command arugments:
```
typedef void lv_obj_t;
```
There are some other changes needed to make the separation of lvgl app in wasm and the native lvgl libray from the runtime.

kisvegabor, 2020-04-21:
I've just responded to the mentioned email address. 

Could you open an issue in the [lvgl repo](https://github.com/littlevgl/lvgl/issues) to describe the required changes?

I'd like to mention that v7.0 will be [released soon](https://github.com/littlevgl/lvgl/issues/1243) so it might be worth using it instead of v6.x. It's available in the `dev-7.0` branch. Basically the library's structure remained the same, however, the "style system" was completely rewritten.

xwang98, 2020-04-22:
@kisvegabor I will file a issue to the lvgl repo for the details. thanks!

closed 2020-04-22 by xwang98

takoyakiyo, 2020-08-13:
May I ask why you used `typedef void lv_obj_t` instead of original definition in `lvgl`?

takoyakiyo, 2020-08-13:
@xwang98 May I ask why you used `typedef void lv_obj_t` instead of original definition in` lvgl`?