wasm-micro-runtime/wasm-micro-runtime #241: Do you have a plan to support riscv architecture ?

zangcq, 2020-04-28:
as above

wenyongh, 2020-05-09:
@zangcq no plan yet, which runtime mode do you want to support, interpreter or AOT? If interpreter, the effort might be relatively small, normally we need to implement the assembly code of the target, e.g. core/iwasm/common/arch/invokeNative_riscv.s, could you please ref to version of other targets? If AOT, we should implement the relocation process, it is relative complex.

zangcq, 2020-05-09:
> @zangcq no plan yet, which runtime mode do you want to support, interpreter or AOT? If interpreter, the effort might be relatively small, normally we need to implement the assembly code of the [...]

I just want to enable the AOT to T-Head C910;
this is a patch to enable LLVM IR bachend for riscv, but i did not verified in simulator.

```
diff --git a/core/iwasm/compilation/aot_llvm.c b/core/iwasm/compilation/aot_llvm.c
index f1776d1..9639e42 100644
--- a/core/iwasm/compilation/aot_llvm.c
+++ b/core/iwasm/compilation/aot_llvm.c
@@ -751,6 +751,7 @@ static ArchItem valid_archs[] = {
     { "x86_64", false },
     { "i386", false },
     { "xtensa", false},
+    { "riscv", false},
     { "mips", true },
     { "aarch64v8", false },
     { "aarch64v8.1", false },
@@ -991,14 +992,20 @@ aot_create_comp_context(AOTCompData *comp_data,

         if (arch) {
             /* Construct target triple: <arch>-<vendor>-<sys>-<abi> */
-            const char *vendor_sys = "-pc-linux-";
-            if (!abi)
-                abi = "gnu";
-            bh_assert(strlen(arch) + strlen(vendor_sys) + strlen(abi) < sizeof(triple_buf));
-            memcpy(triple_buf, arch, strlen(arch));
-            memcpy(triple_buf + strlen(arch), vendor_sys, strlen(vendor_sys));
-            memcpy(triple_buf + strlen(arch) + strlen(vendor_sys), abi, strlen(abi));
-            triple = triple_buf;
+            if (!strcmp(arch, "riscv")) {
+                const char *riscv = "riscv64-unknown-unknown";
+                memcpy(triple_buf, riscv, strlen(riscv));
+                triple = triple_buf;
+            } else {
+                triple = triple_buf;
+                const char *vendor_sys = "-pc-linux-";
+                if (!abi)
+                    abi = "gnu";
+                bh_assert(strlen(arch) + strlen(vendor_sys) + strlen(abi) < sizeof(triple_buf));
+                memcpy(triple_buf, arch, strlen(arch));
+                memcpy(triple_buf + strlen(arch), vendor_sys, strlen(vendor_sys));
+                memcpy(triple_buf + strlen(arch) + strlen(vendor_sys), abi, strlen(abi));
+            }
         }

         if (!cpu && features) {
--

diff --git a/core/iwasm/aot/aot_loader.c b/core/iwasm/aot/aot_loader.c
index f4115bb..136bb8a 100644
--- a/core/iwasm/aot/aot_loader.c
+++ b/core/iwasm/aot/aot_loader.c
@@ -146,6 +146,7 @@ GET_U64_FROM_ADDR(uint32 *addr)
 #define E_MACHINE_MIPS_X   51       /* Stanford MIPS-X */
 #define E_MACHINE_X86_64   62       /* AMD x86-64 architecture */
 #define E_MACHINE_XTENSA   94       /* Tensilica Xtensa Architecture */
+#define E_MACHINE_RISCV64  243      /* RISCV64 Architecture */

 /* Legal values for e_version */
 #define E_VERSION_CURRENT  1        /* Current version */
@@ -206,6 +207,9 @@ get_aot_file_target(AOTTargetInfo *target_info,
         case E_MACHINE_XTENSA:
             machine_type = "xtensa";
             break;
+        case E_MACHINE_RISCV64:
+            machine_type = "riscv";
+            break;
         default:
             if (error_buf)
                 snprintf(error_buf, error_buf_size,
```

wenyongh, 2020-05-12:
That's not enough, maybe wamrc can generate the aot file for riscv target, and the aot file can pass the target check in aot loader, but we should also implement the relocation in aot loader for the aot code, ref to core/iwasm/aot/arch/aot_reloc_arm.c or implementations of other targets.

closed 2020-07-16 by wenyongh