wasm-micro-runtime/wasm-micro-runtime #242: AOT support for iOS platform

keith2018, 2020-04-28:
hi，I have test WAMR on iOS platform, the result shows :

1. the interpreter mode works fine on both iOS simulator and real device
2. AOT mode works fine on simulator (target="x86_64") but fails on real device (target="aarch64")

The test wasm is compiled from simple C code

```c
int test(int n) {
    return n + 42;
}
```

[test.wasm.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/4543059/test.wasm.zip)

```
imports: 
exports:
  [Function] i32 test(i32)
```

AOT compile command
```
./wamrc --target=aarch64 --size-level=0 --opt-level=0 -o test.aot test.wasm
```

Runtime  error

> thread #9, name = 'WebThread', stop reason = EXC_BAD_ACCESS (code=2, address=0x111488004)

[image]

it seems that the func_ptr is invalid

please help,  thanks

xwang98, 2020-04-29:
@keith2018 iOS doesn't support mmap for allocating executable memory, right? Then it is probably not able to support AoT on iOS.

florianwinterstein, 2020-05-06:
> 1. the interpreter mode works fine on both iOS simulator and real device

Keith, would you be so kind and share what you did to get the interpreter to run on iOS?

closed 2020-07-20 by keith2018

Feyewon, 2021-02-18:
@keith2018 would you be so kind and share what you did to get the interpreter to run on iOS?

jjqquu, 2022-05-07:
Have you tried entitlement of allow execution of JIT-compiled code for your app?

iOS applications, which is usually sandboxed,  cannot use mmap’s MAP_JIT without entitlement (the usual way to create a RWX region for JIT purposes). 

https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_cs_allow-jit
https://support.apple.com/fr-mg/guide/security/sec15bfe098e/web

jjqquu, 2022-05-07:
> Have you tried entitlement of allow execution of JIT-compiled code for your app?
>
> iOS applications, which is usually sandboxed, cannot use mmap’s MAP_JIT without entitlement (the usual way to create [...]

Here comes the way Apple suggests to port JIT compiler 
https://developer.apple.com/documentation/apple-silicon/porting-just-in-time-compilers-to-apple-silicon