wasm-micro-runtime/wasm-micro-runtime #244: LLVM Assertion for basic_types->void_ptr_type

TerryGuo, 2020-04-30:
If warm-compiler is built with LLVM that was built in Debug type, then we will get following LLVM assertion when using wamr-compiler:

> $ ./wamrc --target=x86_64  -o test.aot test.wasm
Create AoT compiler with:
  target:        x86_64
  target cpu:
  cpu features:
  opt level:     3
  size level:    3
  output format: AoT file
wamrc: /etc/llvm-build/llvm-10.0.0.src/lib/IR/Type.cpp:639: static llvm::PointerType* llvm::PointerType::get(llvm::Type*, unsigned int): Assertion `isValidElementType(EltTy) && "Invalid type for pointer element!"' failed.
Aborted

Dig this problem with gdb, got below info:

> Program received signal SIGABRT, Aborted.
0x00007ffff0d58277 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install libgcc-4.8.5-39.3.alios7.x86_64 libstdc++-4.8.5-39.3.alios7.x86_64 zlib-1.2.7-16.2.alios7.x86_64
(gdb) bt
#0  0x00007ffff0d58277 in raise () from /lib64/libc.so.6
#1  0x00007ffff0d59968 in abort () from /lib64/libc.so.6
#2  0x00007ffff0d51096 in __assert_fail_base () from /lib64/libc.so.6
#3  0x00007ffff0d51142 in __assert_fail () from /lib64/libc.so.6
#4  0x0000000000606dac in llvm::PointerType::get (EltTy=0x45e3588, AddressSpace=0)
    at /etc/llvm-build/llvm-10.0.0.src/lib/IR/Type.cpp:639
#5  0x0000000000453354 in LLVMPointerType (ElementType=0x45e3588, AddressSpace=0)
    at /etc/llvm-build/llvm-10.0.0.src/lib/IR/Core.cpp:748
#6  0x0000000000435a4c in aot_set_llvm_basic_types (basic_types=0x45e09a8, context=0x45e0ae0)
    at /etc/llvm-build/wasm-micro-runtime/core/iwasm/compilation/aot_llvm.c:691
#7  0x0000000000436dc4 in aot_create_comp_context (comp_data=0x45ddc10, option=0x7fffffffd300)
    at /etc/llvm-build/wasm-micro-runtime/core/iwasm/compilation/aot_llvm.c:1146
#8  0x000000000040f6ac in main (argc=1, argv=0x7fffffffd4d8)
    at /etc/llvm-build/wasm-micro-runtime/wamr-compiler/main.c:178
(gdb)

Seems it is related to 
/etc/llvm-build/wasm-micro-runtime/core/iwasm/compilation/aot_llvm.c:691

Would you please look into it?

TerryGuo, 2020-04-30:
[test.zip](https://github.com/bytecodealliance/wasm-micro-runtime/files/4557165/test.zip)
Here is the test wasm file.

wenyongh, 2020-05-07:
@TerryGuo thanks for finding the issue! I tried to build LLVM with debug type, but got crashed in both Ubuntu 18.04 and Ubuntu 16.04. My command is:

 mkdir -p build
 cd build
 cmake ../llvm \
          -DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
          -DCMAKE_BUILD_TYPE:STRING="Debug" \
          -DLLVM_BUILD_LLVM_DYLIB:BOOL=OFF \
          -DLLVM_OPTIMIZED_TABLEGEN:BOOL=ON \
          -DLLVM_INCLUDE_EXAMPLES:BOOL=OFF \
          -DLLVM_INCLUDE_TESTS:BOOL=OFF \
          -DLLVM_INCLUDE_BENCHMARKS:BOOL=OFF \
          -DLLVM_APPEND_VC_REV:BOOL=OFF
  make

Scanning dependencies of target LTO
[ 87%] Building CXX object tools/lto/CMakeFiles/LTO.dir/lto.cpp.o
[ 87%] Linking CXX shared library ../../lib/libLTO.so
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.

Could you please tell us the build command? thanks a lot.

wenyongh, 2020-05-08:
@TerryGuo, in aot_llvm.c:691, we just define a void pointer type (like C void*) with LLVMPointerType() function, seems that it isn't supported by LLVM debug mode. The void pointer type is used in some translated LLVM IRs, it should be OK to replace them with int8 pointer type, we will fix it soon.

TerryGuo, 2020-05-08:
Hi, it looks to me more like that we define a pointer which points to (void *) pointer.

TerryGuo, 2020-05-08:
@wenyongh the build failure is caused by out of memory. Please reduce the number of build jobs, saying make -j6

wenyongh, 2020-05-08:
@TerryGuo the patch has been merged #250 , could you please help try again?
About the build failure, I tried making with one thread, it also fails, we will try it later.

closed 2020-06-12 by wenyongh