wasm-micro-runtime/wasm-micro-runtime #245: How to manage memory when interacting with AssemblyScript?

hu6360567, 2020-05-05:
I'm trying to write wasm applications by AssemblyScript with native function calls.
Since AssemblyScript builds with a runtime, it handles memory and lifetime.
Then is it safe to use `wasm_runtime_module_malloc` in native functions, while allocating objects in AssemblyScript with its `__alloc` api.

For example, a native function called `getValue` to get value of a string key from hashmap.
The function declaration can be defined as below.
```C
int *getValue(char *key, void *value, size_t length);
void *getValue(char *key);
```
For the first declartion, it may fail when length is not enough.

But for the second declaration, the code use `wasm_runtime_module_malloc` to dynamic allocate memory for `value` based on its size.
**Is it safe to use? Is the allocator aware of memory usage in AssemblyScript runtime?**

If not, what's the best practise for memory management between wamr and wasm?

hu6360567, 2020-05-05:
Another approach is export AssemblyScript's memory alloc/free and call them in native functions.

I'd like to know which one can be the best practise.

xujuntwt95329, 2020-05-09:
Hi,

It's safe to use `wasm_runtime_module_malloc` in your native function as the memory space allocated by this API is from WAMR's `app heap` rather than `wasm linear memory`. Please refer to the picture below:
[image]

The `wasm linear memory` is used by wasm itself, for example Assemblyscript builds tlsf into its runtime, and it provides `__alloc` API to allocate memory from `wasm linear memory`. If `__alloc` API is exported, you can call it to allocate a memory space, and its managed by Assemblyscript's runtime. 

If '__alloc' API is not exported, or in some cases there is no mechanism to allocate memory in `wasm linear memory`, then `app heap` can be used, it's another memory space besides the `wasm linear memory`, and will be treated as valid memory space in WAMR.

For your question:
> Is it safe to use? Is the allocator aware of memory usage in AssemblyScript runtime?

Yes, it is safe to use  `wasm_runtime_module_malloc`, it will not influence the memory managed by Assemblyscript runtime, and WAMR know nothing about the memory usage in AssemblyScript runtime.

> Another approach is to use AssemblyScript's exported memory alloc/free and call them in native functions.
I'd like to know which one can be the best practise.

Both approaches will work well for your case, but there may be some difference:
1. The size of WAMR's `app_heap` is fixed, you can use `--heap-size = `  to define how much heap space is needed. So if the `app_heap` is not enough, you can't get a memory space by `wasm_runtime_module_malloc`; The size of `wasm linear memory` may be changeable by `memory.grow` opcode, if you use `__alloc`, it may grow the memory space if the space in current page is not enough. The maximum page is defined in wasm bytecode, it may be configurable by `asc` compiler
2. We don't have GC in `app heap`, so you need to manage the memory you allocated by `wasm_runtime_module_malloc`. If you need GC for the memory you allocated, you need to use `__alloc` API provided by as's runtime and use some other GC related APIs like `__retain`, `__release` and so on.

I'm also interested in AssemblyScript, and I'm glad to know that your AssemblyScript applications can run on WAMR, please feel free to raise an issue if you meet another problem, or you can also discuss more details about WAMR with me through email `693788454@qq.com`

Thanks

rahul-thakoor, 2020-05-14:
hey @xujuntwt95329 
thanks for the explanation above. I'm also interested in using AssemblyScript with WAMR along with WASI support.

Have you been able to use [as-wasi](https://github.com/jedisct1/as-wasi) ? I ran into this issue https://github.com/jedisct1/as-wasi/issues/32

Woud really appreciate some help. 🙏

xujuntwt95329, 2020-05-15:
Could you please provide some test cases ? For example your test code, your environment string.

I've tried the example code from `as-wasi`, but can't reproduce the issue you mentioned.
``` typescript
import { Console, Environ } from "../node_modules/as-wasi/assembly";

let env = new Environ();
let home = env.get("HOME")!;
Console.log(home);
```

``` bash
iwasm --env="HOME=test" as-wasi.wasm
```

and I got this in my console
``` bash
test
```

Seems it works well

rahul-thakoor, 2020-05-18:
you are right! it seems to work. must have been an error on my part.

Thanks a lot!

closed 2020-09-03 by wenyongh