wasm-micro-runtime/wasm-micro-runtime #246: About AOT memory access boundary check

linzhiping, 2020-05-07:
> in /core/iwasm/aot/aot_runtime.c:
`bool
aot_validate_app_addr(AOTModuleInstance *module_inst,
                      int32 app_offset, uint32 size)
{
    /* integer overflow check */
    if(app_offset + (int32)size < app_offset) {
        goto fail;
    }
    // Q:
   //  should it be like this?
   // if (app_offset < module_inst->heap_base_offset
   //     || app_offset + (int32)size >= (int32)module_inst->memory_data_size) {
   //     goto fail;
   // }
   // since app_offset == module_inst->heap_base_offset is valid
   // app_offset + (int32)size == (int32)module_inst->memory_data_size) is invalid
    if (app_offset <= module_inst->heap_base_offset
        || app_offset + (int32)size > (int32)module_inst->memory_data_size) {
        goto fail;
    }
...
}
void *
aot_addr_app_to_native(AOTModuleInstance *module_inst, int32 app_offset)
{
    int32 memory_data_size = (int32)module_inst->memory_data_size;
    // Q:
    // should it be like this?
    // if (module_inst->heap_base_offset <= app_offset
    //    && app_offset < memory_data_size)
    if (module_inst->heap_base_offset < app_offset
        && app_offset < memory_data_size)
        return (uint8*)module_inst->memory_data.ptr + app_offset;
    return NULL;
}
`

wenyongh, 2020-05-08:
@linzhiping thanks a lot for finding the issue! It is really an issue, and some other codes have the same problem, we will fix them soon.

wenyongh, 2020-05-08:
@linzhiping the patch has been merged, [#249](https://github.com/bytecodealliance/wasm-micro-runtime/pull/249). Note for second condtion: app_offset + (int32)size > (int32)module_inst->memory_data_size), equal is allowed: suppose that memory_data_size is 65536, app_offset is 65535 and size is 1, it is OK to read one byte from offset 65535, and here app_offset + size == memory_data_size.

closed 2020-05-15 by wenyongh