wasm-micro-runtime/wasm-micro-runtime #247: wasmtime-micro-runtime in Rust

redradist, 2020-05-07:
Consider to rewrite wasm-micro-runtime on Rust with name wasmtime-micro-runtime !!
Reason, because Rust provide more safer implementation and at the same time it is still as much powerful as C or C++

I beleave, it is possible to get part of wasmtime implementation and reduce it to fit in wasmtime-micro-runtime requirements ...

I could try to help with writing it ...

rahul-thakoor, 2020-05-12:
I was looking for  Rust bindings as a start here #230 

Are you able to help with that @redradist?

redradist, 2020-05-12:
@redradist 
> 
> I was looking for Rust bindings as a start here #230
> 
> Are you able to help with that @redradist?

Yeah, I think I can find some time for Rust Bindings ...

Do you have some work done with it ?

redradist, 2020-05-12:
> 
> 
> I was looking for Rust bindings as a start here #230
> 
> Are you able to help with that @redradist?

rahul-thakoor, 2020-05-12:
Not really. I'm too familiar with bindgen tbh

closed 2020-08-28 by wenyongh