wasm-micro-runtime/wasm-micro-runtime #251: AOT wasm buffer has implicit 32bit alignment requirement

exFalso, 2020-05-08:
Currently the AOT loader has a requirement that the passed in buffer must be 32bit aligned. The code returns an opaque `magic header not detected` error if this is not the case. The alignment requirement is implicit in the use of `read_uint32` at https://github.com/bytecodealliance/wasm-micro-runtime/blob/7abd1ca813794db267aeb9fb18d465f461a0d4da/core/iwasm/aot/aot_loader.c#L1816 which re-aligns the buffer pointer thus offsetting the magic header location if the buffer is not properly aligned.

I think this requirement should either be dropped or documented with three exclamation points.

wenyongh, 2020-05-09:
@exFalso yes, the AOT buffer must be 4-byte aligned in current design. We will update the the related document. Thanks a lot!

wenyongh, 2020-05-15:
Comments of API wasm_runtime_load() was updated, and seems that we can close the issue.

closed 2020-05-15 by wenyongh