wasm-micro-runtime/wasm-micro-runtime #253: not portable posix api "pthread_getattr_np"

hu6360567, 2020-05-12:
when porting to other posix compatible platform, the posix module may not be compiled due to the `os_thread_get_stack_boundary`  implementation, in which `pthread_getattr_np` is a GNU extension.

Is there any plan to replace `pthread_getattr_np`?

wenyongh, 2020-05-12:
Yes, we may enhance the implementation: use pthread_getattr_np() for linux, and use pthread_get_stackaddr_np() for MacOS. Not sure what is your platform, and is there similar API to use?

hu6360567, 2020-05-12:
My target platform has a ported version of `musl-1.1.22/libc` with partial pthread support of posix.

Unfortunately, there is no simliar api to use, but there are apis like `pthread_attr_setstack` and `pthread_attr_setstacksize`.
Is there any way to implement stack boundary when the thread is created with apis above?

hu6360567, 2020-05-12:
The key problem is `os_thread_get_stack_boundary` can get `pthread_t` but there is no apis to get `pthread_attr_t` from `pthread_t`.
Another approach is implementing a hashtable for `<pthread_t, stack_addr>` and wrapping an invoking api to create a separate thread, which saves the `stack_addr` of this separate thread into that table, so that `os_thread_get_stack_boundary` can lookup the `stack_addr` on the go.

closed 2020-05-12 by hu6360567