wasm-micro-runtime/wasm-micro-runtime #254: Signal 4 crash on MIPS 32 devices on FAST_INTERP

AlexEne, 2020-05-12:
Hello,

I keep having this strange crash that I encountering on a MIPS32 device I was playing with.
The reference (not fast) interp works perfectly fine.
However, when trying to build a program that just does a simple vector allocation and adds elements in a vector, it crashes with a signal 4 if built with the WAMR_BUILD_INTERP set to 1.

I've narrowed the crash down to the following place in `wasm_interp_fast.c` in `call_func_from_interp`.

The offender is the last line of this line:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/core/iwasm/interpreter/wasm_interp_fast.c#L2386

```c
  call_func_from_interp:
      // Stuff happens here 
      else {
          *(outs_area->lp) = GET_OPERAND(int32, (2 * (cur_func->param_count - i - 1)));;
          outs_area->lp ++;
        }
      }
```

I did try my best to figure out what's happening, but the fields around `cur_func`, `outs_area` are a bit unclear. In addition to that, I can't find any documentation on what exactly fast_interp does differently from the reference interpreter. 
The program (or more complex programs) work just fine if I disable the fast interpreter.

For reference here are the CMAKE settings that __do not__ work:
```cmake
set (WAMR_BUILD_PLATFORM "linux")
# set (WAMR_BUILD_TARGET X86_64)
set (WAMR_BUILD_JIT 0)
set (WAMR_BUILD_AOT 0)
set (WAMR_BUILD_INTERP 1)
set (WAMR_BUILD_FAST_INTERP 1)
set (WAMR_BUILD_LIBC_BUILTIN 0)
set (WAMR_BUILD_LIBC_WASI 1)
set (WAMR_BUILD_TARGET "MIPS")
```

Setting `WAMR_BUILD_INTERP to 0` makes everything work as intended.

The program I used as a minimal example to trigger this crash:
```rust
#[no_mangle]
pub extern "C" fn main() {
    for i in 0..10 {
        a.push(i as f32);
    }
}
```

I've built it with `cargo build --release --target wasm32-wasi`

Rochet2, 2020-05-13:
It seems that the minimal example is missing the definition of `a` and cannot be compiled as is.

AlexEne, 2020-05-13:
I am fairly certain the casts aren't needed and it just works with a `Vec<i32>` and `a.push(i)`, but I've copied the last thing I tried as a minimal example.

xujuntwt95329, 2020-05-15:
Hi @AlexEne, thanks for reporting this issue !

We have tried your case on `x86_64` and `qemu_system_mips`, but we can't reproduce this problem. So I guess this issue may be hardware related, for example the `alignment requirement`, `endian`, `FPU` and so on.

I'll try to test on a real mips device later, can you provide some detailed information about your CPU ? For example the target ABI, `mips` or `mipsel`. Also could you provide some compiled wasm files to test ?

> I've fixed the example, but am fairly certain the casts aren't needed and it just works with a Vec<i32> and a.push(i). As a precaution measure, I've copied the last thing I tried as a minimal example and generated this bug.

Seems this problem is related to float-point, are you using `-msoft-float` or `-mhard-float` when compiling WAMR ?

Thanks.

AlexEne, 2020-05-15:
I am going to check and get back to you.

AlexEne, 2020-05-15:
My CPU is:
```
system type     : MStar Chip
processor       : 0
cpu model       : MIPS 74Kc V5.5  FPU V0.0
BogoMIPS        : 449.74
isa         : mips1 mips2 mips32r1 mips32r2
``` 

I can also reproduce the issue with the following rust program (so it's not float related):
```rust
#[no_mangle]
pub extern "C" fn main() {
    let mut a: Vec<i32>  = Vec::new();
    for i in 0..10 {
        a.push(i as i32);
    }
}
```

readelf on the host program output is:
```
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           MIPS R3000
  Version:                           0x1
  Flags:                             0x70001005, noreorder, cpic, o32, mips32r2
```

closed 2021-02-23 by wenyongh