wasm-micro-runtime/wasm-micro-runtime #255: 'pthread_getattr_np' not available on MacOS

keith2018, 2020-05-15:
Hi,  when I try to build WAMR on MacOS

```
set (WAMR_BUILD_PLATFORM "darwin")
set (WAMR_BUILD_TARGET "X86_64")
set (WAMR_BUILD_INTERP 1)
set (WAMR_BUILD_FAST_INTERP 1)
set (WAMR_BUILD_AOT 1)
set (WAMR_BUILD_JIT 0)
set (WAMR_BUILD_LIBC_BUILTIN 1)
set (WAMR_BUILD_LIBC_WASI 0)
```

it failed with error :
```
[ 47%] Linking C shared library libiwasm.dylib
Undefined symbols for architecture x86_64:
  "_pthread_getattr_np", referenced from:
      _os_thread_get_stack_boundary in darwin_thread.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libiwasm.dylib] Error 1
make[1]: *** [CMakeFiles/libiwasm.dir/all] Error 2
make: *** [all] Error 2
```

https://github.com/bytecodealliance/wasm-micro-runtime/blob/1362b6d81f9045be0325287ad99cc0ed064ee9c6/core/shared/platform/common/posix/posix_thread.c#L230

maybe 'pthread_getattr_np' is not available on MacOS ?

wenyongh, 2020-05-15:
Thanks a lot @keith2018 , we have found the issue #253, and have fixed it in patch #256 , could you please pull the latest code and test again? thanks.

keith2018, 2020-05-19:
> Thanks a lot @keith2018 , we have found the issue #253, and have fixed it in patch #256 , could you please pull the latest code and test again? thanks.

it works fine now, thanks

closed 2020-05-21 by wenyongh