wasm-micro-runtime/wasm-micro-runtime #260: Questions while reading the codes of app-framework

oubotong, 2020-05-22:
When I first read the code in the `samples/simple/iwasm_main.c`, I find out it will call
 ```
init_sensor_framwork(){
...
wasm_register_msg_callback(SENSOR_EVENT_WASM, app_mgr_sensor_event_callback);
...
}
```
to initialize the **sensor** and set up the **callback**. 
Then in the callback function`app_mgr_sensor_event_callback()`, it will lookup the real handler function we need to callback:
```
app_mgr_sensor_event_callback(){
...
func_onSensorEvent = wasm_runtime_lookup_function(inst, "on_sensor_event", "(i32i32i32)");
...
}
```
However, for the function with name `on_sensor_event()` and **i32i32i32** signature is the following located here `app-framework/sensor/app/sensor.c`:
```
void on_sensor_event(uint32 sensor_id, char * buffer, int len)
{
    attr_container_t * sensor_data = (attr_container_t *) buffer;

    // lookup the sensor and call the handlers
    sensor_t s = g_sensors;
    sensor_t prev = NULL;
    while (s) {
        if (s->handle == sensor_id) {
            s->sensor_callback(s, sensor_data, s->user_data);
            break;
        }

        s = s->next;
    }
}
```
My question is that I see another `sensor.c` file located in ``samples/simple/wasm-apps.sensor.c` which contains a function:
```
/* Sensor event callback*/
void sensor_event_handler(sensor_t sensor, attr_container_t *event,
                          void *user_data)
{
    printf("### app get sensor event\n");
    attr_container_dump(event);
}
```
How does this function works since it isn't set up as the call back function anywhere in the app-framework as I see. Can anyone tell me how is the function above been set up as the call back function?

xwang98, 2020-05-23:
@oubotong  The samples/simple/wasm-apps.sensor.c is the source file which will be compiled to WASM bytecode. It call the following API to open the sensor "sensor_test" with callback function sensor_event_handler for handling all events on this sensor.
```
sensor = sensor_open("sensor_test", 0, sensor_event_handler, user_data);
```
You may want to refer to the source file for how sensor_open() works:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/core/app-framework/sensor/app/sensor.c

This diagram should be helpful for understanding the working flow:
https://github.com/bytecodealliance/wasm-micro-runtime/blob/master/doc/pics/sensor_callflow.PNG

oubotong, 2020-05-23:
Thanks for your help! I have figured it out.

closed 2020-05-23 by oubotong