wasm-micro-runtime/wasm-micro-runtime #261: wamr-compiler compilation keeps failing in MacOS

dpinthinker, 2020-05-23:
After executing `build_llvm.sh`, execute the following commands:

```
mkdir build
cd build
cmake ..
build
```

then it fails.

```
➜  build git:(master) make
Scanning dependencies of target vmlib
[  2%] Building C object CMakeFiles/vmlib.dir/Users/zhujinpeng/workspace/webasm/wasm-micro-runtime/core/shared/platform/linux/platform_init.c.o
clang: error: unknown argument: '-mindirect-branch-register'
clang: warning: -Wl,-z,noexecstack,-z,relro,-z,now: 'linker' input unused [-Wunused-command-line-argument]
make[2]: *** [CMakeFiles/vmlib.dir/Users/zhujinpeng/workspace/webasm/wasm-micro-runtime/core/shared/platform/linux/platform_init.c.o] Error 1
make[1]: *** [CMakeFiles/vmlib.dir/all] Error 2
make: *** [all] Error 2
```

liurenju, 2020-05-23:
How did you configure the building environment? By the way, you should try to build the runtime on Linux since the one you used was supposed to be used in linux.

dpinthinker, 2020-05-23:
> How did you configure the building environment? By the way, you should try to build the runtime on Linux since the one you used was supposed to be used in linux.

Yes, I build it successfully on my Linux cloud machine. But I still want to build it on my MacOS laptop. Could you tell me how to build it on MacOS?

wenyongh, 2020-05-23:
@dpinthinker , could you please comment the -mindirect-branch-register flag in wamr-compiler/CMakeLists.txt, and build again?

    if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
    set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
    endif ()

to

    if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
    #set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
    endif ()

dpinthinker, 2020-05-23:
After comment the -mindirect-branch-register flag, this error is gone.
But another issue come out.

```
[ 50%] Building ASM object CMakeFiles/vmlib.dir/Users/zhujinpeng/workspace/webasm/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s.o
/Users/zhujinpeng/workspace/webasm/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s:9:5: error: unknown directive
    .type invokeNative, @function
    ^
make[2]: *** [CMakeFiles/vmlib.dir/Users/zhujinpeng/workspace/webasm/wasm-micro-runtime/core/iwasm/common/arch/invokeNative_em64.s.o] Error 1
make[1]: *** [CMakeFiles/vmlib.dir/all] Error 2
make: *** [all] Error 2
```

wenyongh, 2020-05-23:
Yes, could you please try modifying:

    .globl invokeNative
         .type    invokeNative, @function
to

    .globl _invokeNative
and build again?

oubotong, 2020-05-23:
Thanks for your help! I have figured it out

dpinthinker, 2020-05-24:
> Yes, could you please try modifying:
> 
> ```
> .globl invokeNative
>      .type    invokeNative, @function
> ```
> 
> to
> 
> ```
> .globl _invokeNative
> ```
> 
> and build again?

.type error is gone. Another ld error.

```
[100%] Linking CXX executable wamrc
ld: unknown option: --gc-sections
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wamrc] Error 1
make[1]: *** [CMakeFiles/wamrc.dir/all] Error 2
make: *** [all] Error 2
```

dpinthinker, 2020-05-24:
After I commented this line in CMakeList.txt
```
set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--gc-sections")
```
It still output an error.
```
[100%] Linking CXX executable wamrc
Undefined symbols for architecture x86_64:
  "_invokeNative", referenced from:
      _wasm_runtime_invoke_native in libvmlib.a(wasm_runtime_common.c.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wamrc] Error 1
make[1]: *** [CMakeFiles/wamrc.dir/all] Error 2
make: *** [all] Error 2
```

wenyongh, 2020-05-24:
Could you try modifying  ".globl _invokeNative" to ".globl invokeNative"?

dpinthinker, 2020-05-24:
sting not working

dpinthinker, 2020-05-24:
Here is all my changes.
```
diff --git a/core/iwasm/common/arch/invokeNative_em64.s b/core/iwasm/common/arch/invokeNative_em64.s
index 739e84e..b975e71 100644
--- a/core/iwasm/common/arch/invokeNative_em64.s
+++ b/core/iwasm/common/arch/invokeNative_em64.s
@@ -6,7 +6,7 @@
     .align 2
 #ifndef BH_PLATFORM_DARWIN
 .globl invokeNative
-    .type    invokeNative, @function
+#    .type    invokeNative, @function
 invokeNative:
 #else
 .globl _invokeNative
diff --git a/wamr-compiler/CMakeLists.txt b/wamr-compiler/CMakeLists.txt
index 336e317..b43a009 100644
--- a/wamr-compiler/CMakeLists.txt
+++ b/wamr-compiler/CMakeLists.txt
@@ -76,7 +76,7 @@ add_definitions(${LLVM_DEFINITIONS})
 message(STATUS "Found LLVM ${LLVM_PACKAGE_VERSION}")
 message(STATUS "Using LLVMConfig.cmake in: ${LLVM_DIR}")

-set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--gc-sections")
+#set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--gc-sections")
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ffunction-sections -fdata-sections \
                     -Wall -Wno-unused-parameter -Wno-pedantic")

@@ -102,7 +102,7 @@ include (${IWASM_DIR}/compilation/iwasm_compl.cmake)
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Wextra -Wformat -Wformat-security")
 # set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wconversion -Wsign-conversion")
 if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
-set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
+#set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
 endif ()
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fstack-protector-strong --param ssp-buffer-size=4")
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wl,-z,noexecstack,-z,relro,-z,now")
```

wenyongh, 2020-05-24:
Hi, the codes of #else branch should work for MacOS, we are able to build iwasm of platform darwin in MacOS (product-mini/platforms/darwin), in which BH_PLATFORM_DARWIN is defined:

#ifndef BH_PLATFORM_DARWIN
.globl invokeNative
    .type    invokeNative, @function
invokeNative:
#else
**.globl _invokeNative
_invokeNative:**
#endif /* end of BH_PLATFORM_DARWIN */

So could you please modify these codes to
**.globl _invokeNative
_invokeNative:**

Or modify
#ifndef BH_PLATFORM_DARWIN
 to
#if defined(BH_PLATFORM_DARWIN) || defined(__APPLE__)
to let #else branch take effect and try again?

dpinthinker, 2020-05-24:
@wenyongh Thank you so much! It is finally compiled successfully.

Follow is the changes.
```
diff --git a/wamr-compiler/CMakeLists.txt b/wamr-compiler/CMakeLists.txt
index 336e317..dcaa98a 100644
--- a/wamr-compiler/CMakeLists.txt
+++ b/wamr-compiler/CMakeLists.txt
@@ -4,7 +4,7 @@ cmake_minimum_required (VERSION 2.8)

 project (aot-compiler)

-set (WAMR_BUILD_PLATFORM "linux")
+set (WAMR_BUILD_PLATFORM "darwin")

 # Reset default linker flags
 set (CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
@@ -76,7 +76,7 @@ add_definitions(${LLVM_DEFINITIONS})
 message(STATUS "Found LLVM ${LLVM_PACKAGE_VERSION}")
 message(STATUS "Using LLVMConfig.cmake in: ${LLVM_DIR}")

-set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--gc-sections")
+#set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--gc-sections")
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ffunction-sections -fdata-sections \
                     -Wall -Wno-unused-parameter -Wno-pedantic")

@@ -102,7 +102,7 @@ include (${IWASM_DIR}/compilation/iwasm_compl.cmake)
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Wextra -Wformat -Wformat-security")
 # set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wconversion -Wsign-conversion")
 if (WAMR_BUILD_TARGET MATCHES "X86_.*" OR WAMR_BUILD_TARGET STREQUAL "AMD_64")
-set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
+#set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -mindirect-branch-register")
 endif ()
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fstack-protector-strong --param ssp-buffer-size=4")
 set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wl,-z,noexecstack,-z,relro,-z,now")
```

dpinthinker, 2020-05-28:
Fixed by #265

closed 2020-05-28 by dpinthinker