wasm-micro-runtime/wasm-micro-runtime #262: Does WAMR support sharing memory between modules?

oubotong, 2020-05-23:
I read the proposal from the meeting of the WebAssembly group:
[Multiple Memories for Wasm](https://github.com/WebAssembly/multi-memory/blob/master/proposals/multi-memory/Overview.md) 

It seems that the [Linear Memory Index Space of a module](https://webassembly.org/docs/modules/) can maintain the index of imported external memory space. I am wondering if WARM has supported this because from issue #92 it seems currently every module can only access its own memory space and a private app heap.

Rochet2, 2020-05-24:
Based on the issue you linked as well as https://github.com/bytecodealliance/wasm-micro-runtime/issues/190#issuecomment-597963563 and https://github.com/bytecodealliance/wasm-micro-runtime/issues/200 there is no sharing capabilities yet by default.

However, you can instantiate multiple modules and copy parts of memory from one to the other, which has worked for me.

[WAMR also seems to support defining your own malloc, realloc and free functions](https://github.com/bytecodealliance/wasm-micro-runtime/blob/f1a0e75ab78337bce13a02d7f02ee1b22cf06823/product-mini/platforms/linux/main.c#L252-L255). Using this you could make allocation functions that use mmap to create virtual memory and possibly during runtime then define the shared areas of the memory for the modules. This is something I have yet to test.

xwang98, 2020-05-24:
@oubotong the pthread supporting is on the way. It will allow instances share single linear space. Can it solve what you need?

oubotong, 2020-05-25:
Yes, it could solve my problem but I am also thinking of how to keep a good isolation while sharing a linear memory. I am thinking if we can share the memory during runtime, like when a new module initiates, he can join in the existing sharing memory space to utilize some results computed by other modules or do some acceleration stuff to the existing task of other modules. We may need some authorization methods during runtime to achieve this (just my personal thoughts).

Another thing I want to question is that WASM says it uses signatures to verify whether your intentional function call is valid in the function table which is basically in a format like `i32i32i32` , what if the attacker just crafts his adversarial function to have the same type of parameters and return values as the `i32i32i32` to make it valid. 
I may misunderstand something here, please tell me if I make any mistake.

Rochet2, 2020-05-25:
According to https://webassembly.org/docs/security/ Clang/LLVM is able to provide some additional checks for CFI with `-fsanitize=cfi`, the page also touches on coming additions reagarding CFI.

closed 2020-12-16 by wenyongh